dotintent/react-native-ble-plx #791: Does device scan work when phone is locked?

qiray, 2020-12-22:
Hello!

I use react-native-ble-plx to scan devices and it works well while the phone isn't locked. When I lock the phone the scan stops. After device is unlocked scan works again. 

So is it possible to scan BLE devices while the phone is locked?

I use LowLatency mode for scanning
``` js
{ scanMode: ScanMode.LowLatency }
```

dariuszseweryn, 2020-12-22:
It is but your app needs to have proper permission granted by the user (to use always) and you need to use scan filters.

qiray, 2020-12-23:
@dariuszseweryn, I added android.permission.ACCESS_BACKGROUND_LOCATION to AndroidManifest.xml and enabled "Allow all the time" in settings but it seems to have no effect. What filters and permissions should I add to make it work?

dariuszseweryn, 2020-12-23:
Filters adequate for the advertisement of the peripheral you are interested it. You cannot "scan everything" in the background because it may be used to locate the user.

qiray, 2020-12-23:
So I need to add UUIDs list for correct background scan, right?

dariuszseweryn, 2020-12-23:
Right

closed 2021-06-07 by dariuszseweryn