dotintent/react-native-ble-plx #797: onDisconnected callback taking more than 10 seconds to identify on disconnect of device in android

Priyanka-SP, 2021-01-10:
onDisconnected callback is getting triggered after 10 seconds in android but works fine in ios which gets triggered as soon as the Bluetooth device gets turned off.
Here is how Im using this callback method
 device
        .connect()
        .then((dev) => {
          dev.onDisconnected((error,devs) => {
           alert('device disconnected')
          });
          return dev.discoverAllServicesAndCharacteristics();
        })

ThomasStubbe, 2021-01-26:
Same issue here (Android, version 2.0.2). And if the device is back online within 10 seconds, it will **never send** the **onDisconnect**, although the **device is disconnected**

LackoDan, 2021-01-26:
Another +1 from me. Because of this issue, we cannot correctly update our UI state on Android anymore on disconnect. Please fix this as soon as possible.

holakmateusz, 2021-03-04:
Same issue we have on Android, newest version of library.

closed 2021-06-11 by stale[bot]