dotintent/react-native-ble-plx #798: Android: Unable to use writeCharacteristicWithResponseForService function

SOLR4189, 2021-01-12:
Hello,

I am using I am using this library to connect a React Native app (Android) to a BLE device (raspberry pi). On calling  `writeCharacteristicWithResponseForService ` function I get next error:

```
{
	"message": "Characteristic YYYYYYYYYYYY write failed for device XXXXXXXXXXXXX and service ZZZZZZZZZZZZZZZ",
	"errorCode": 401,
	"attErrorCode": 14,
	"iosErrorCode": null,
	"androidErrorCode": null,
	"reason": "GATT exception from MAC address XXXXXXXXXXXXX,
	status 14 (GATT_ERR_UNLIKELY),
	type BleGattOperation{description='CHARACTERISTIC_WRITE'}.
	(Look up status 0x0e here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)",
	"name": "BleError",
	"line": 24736,
	"column": 28,
	"sourceURL": "http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"
}
```

My code:

```
scanAndConnect() {
	this.manager.startDeviceScan(null, null, (error, device) => {			 	  
		if (error) {
			this.error(error.message)
			return
		}
		
		if (deviceName === 'MyDevice') {
			this.manager.stopDeviceScan()
						
			let connectedDevice = async () => {
				const conDevice = await this.manager.connectToDevice(device.id)
				const services = await conDevice.discoverAllServicesAndCharacteristics()
			};
			
			connectedDevice()
				.then(() => {
					device.services().then(services => {
						this.manager.isDeviceConnected(device.id).then(console.log) //true

						device.writeCharacteristicWithResponseForService('ZZZZZZZZZZZZZZZZZZ',
								'YYYYYYYYYYYYYYYYY', base64.encode("Hello World")))
							.then((characteristic) => {
								this.info(base64.decode(characteristic.value));
								return 
							})
							.catch((error) => {
								console.log(JSON.stringify(error))
							})
					})
				})
		}
	});
}
```

Any ideas? 

Thanks

closed 2021-01-13 by SOLR4189

existdissolve, 2021-01-19:
@SOLR4189 Did you resolve this? If so, what was the issue and resolution? Thanks!

SOLR4189, 2021-01-19:
@existdissolve Yes. I restarted my phone and raspberry pi, then reinstalled application on my phone.