dotintent/react-native-ble-plx #799: Deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 8.0

ashok-sl, 2021-01-13:
React native Showing Recent Messages The iOS Simulator deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 8.0, but the range of supported deployment target versions is 9.0 to 14.3.99.

RN Version - 0.63.3
Platform - IOS
xCode Version - 12.3

Can I send and receive a nearby beacon with this package?

copiri-six, 2021-01-13:
+1.  If I understand this error correctly, it seems that xCode no longer allows this library to be integrated because the library targets an iOS version that is lower than the minimum supported version of xCode.  Seems like this means that the library can no longer be used, if you are using the latest version of xCode.  Which means either I have to downgrade my xCode version (ouch) or find another library to support my BLE tasks (bigger ouch).

Is this as simple as changing [this line](https://github.com/Polidea/react-native-ble-plx/blob/master/ios/Podfile#L2) from `platform :ios, '8.0'` to `platform :ios, '9.0'`?

Or... maybe I'm missing something?

closed 2021-06-11 by stale[bot]

suprith-s-reddy, 2021-06-20:
I have the same issue.