dotintent/react-native-ble-plx #800: Bug Report

MarcFarres, 2021-01-15:
- [ * ] I am running the latest version
- [ * ] I checked the documentation and found no answer
- [ * ] I checked to make sure that this issue has not already been filed

When i build an android project with react-native with"npx react-native run android" i have the following error:

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
(  ...  )
1: Task failed with an exception.

Execution failed for task ':react-native-ble-plx:processDebugManifest'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > java.io.FileNotFoundException:  ( ... )\node_modules\react-native-ble-plx\android\build\intermediates\manifest_merge_blame_file\debug\manifest-merger-blame-debug-report.txt (Acceso denegado)

2: Task failed with an exception.

Execution failed for task ':react-native-ble-plx:parseDebugLibraryResources'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
  \node_modules\react-native-ble-plx\android\build\intermediates\local_only_symbol_list\debug\parseDebugLibraryResources\R-def.txt

I tried to remove and regenrate build folder with same result. Error occurs since i upgraded gradle.

dariuszseweryn, 2021-01-15:
Could you try and build the app by going into `/android` and typing `./gradlew assembleDebug`?

dariuszseweryn, 2021-04-14:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

closed 2021-04-14 by dariuszseweryn

VenturitPiyush, 2022-07-08:
Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.Polidea:MultiPlatformBleAdapter:58c87b49f12f4a5ab6f7af31f8085249e206400a.