dotintent/react-native-ble-plx #801: Bug Report

bigdreamer07, 2021-01-17:
I am trying to get notification from Kotlin on React Native using BLE (Bluetooth Low Energy) connection. Devices are connected each other and I am trying to send notification from Android device to iPhone device. I am using react-native-ble-plx in React Native and BluetoothGatt in Kotlin. Service for notification is added exactly, but I am always getting this type of error.

> [BleError: Characteristic 00002b01-0000-1000-8000-00805f9b34fb notify change failed for device 7DE9469A-26EA-472F-97D7-16786143626A and service 00002b00-0000-1000-8000-00805f9b34fb]

I was not able to find solution during several days of researching by the internet. Below is my React Native side's code.

```
let service = '00002B00-0000-1000-8000-00805F9B34FB';
    let characteristic = '00002B01-0000-1000-8000-00805F9B34FB';
    const sub = KikiBleManager.bleManager.monitorCharacteristicForDevice(connectedDevice.id, service, characteristic, function (err, result) {
  if(err) {
    console.log(err)
    return;
  }
  console.log(result);
});
```

dariuszseweryn, 2021-04-14:
Have you added a Client Characteristic Configuration Descriptor on the peripheral side?

CodeLogist, 2021-04-16:
Hey, 

It seems like both the service UUID and the characteristic UUID of your service are reserved ones.
Please refer this: https://www.slideshare.net/StanleyChang13/ble-overview-andimplementation and find 00002B00-0000-1000-8000-00805F9B34FB UUID there. It seems like it's reserved for MicroAppManager Service UUID and the characteristic UUID is reserved for MicroApp Installer Characteristic. 

So, just change the UUIDs and add the changes @dariuszseweryn suggested.

mciechanowicz, 2022-01-04:
HI.
I think the previous comments are sufficient answer, if the problem persists please open a new issue. Greetings!

closed 2022-01-04 by mciechanowicz