dotintent/react-native-ble-plx #803: Question: manufacturerData null but device send hex data

livindead, 2021-01-18:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

BLE device send ASC II  HEX data and another scanner app can read from manufacturerData data (0x080931020230303030) but startDeviceScan return null in manufacturerData

code example
`
 this.manager.startDeviceScan(null, null, (error, device) => {
      if (error) { console.log("error", error)return }
      console.log( "device ", device.name, device.manufacturerData  );
})
`

dariuszseweryn, 2021-01-19:
And what is the question? What have you tested on? What is the device you try to scan? OS? Version? Anything?

livindead, 2021-01-19:
os: android 10, custom device on Microchip.
why library return null ?
what type of data  must send device:  string, hex , bin ?

dariuszseweryn, 2021-01-19:
> why library return null ?

I do not know. You have not included native logs with debugging turned on. That could help with knowing what the OS received.

> what type of data must send device: string, hex , bin ?

Does not matter. BLE sends bytes after all.

Bikash-Sawa, 2021-03-07:
Same here i get  device.manufacturerData is null. In android version 10

josuegomez, 2021-03-28:
manufacturerData null in IOS and Android 10 too

dariuszseweryn, 2021-04-06:
This issue has no actionable info. If the manufacturer's data is null then either the peripheral does not advertise it or the OS does not deliver the data to the app. Impossible to say without logs. Closing.

closed 2021-04-06 by dariuszseweryn