dotintent/react-native-ble-plx #806: Bug Report: Disconnecting when subscriptions removed or transactions cancelled

amduffy19, 2021-01-21:
## Expected Behavior

Should be able to cancel all current subscriptions with `subscription.remove()` or `bleManager.cancelTransaction(transactionID)` and the device should stay connected and allow for future reads/writes/monitoring.

## Current Behavior

If all subscriptions are cancelled, the device disconnects after 20-30 seconds. It seems to be fine as long as at least one subscription is still monitoring.

### Steps to Reproduce

1. Connect to a device
2. Call `device.monitorCharacteristicForService`
3. Call `remove()` on the subscription or `bleManager.cancelTransaction(transactionID)`
4. Wait 20-30 seconds
5. Device will disconnect

### Context

* Library version: 2.0.1
* Platform: At least Android (haven't tested on iOS)

amduffy19, 2021-02-04:
Downgrading to 2.0.1 seems to have fixed it for now.

dariuszseweryn, 2021-04-14:
You have not added any logs. I expect that your peripheral crashes when canceling characteristic monitoring and after 20 seconds the supervision timeout kicks in.

closed 2021-04-14 by dariuszseweryn