dotintent/react-native-ble-plx #808: Trying to read device when scanning just gives an object full of errors

littleman27, 2021-01-25:
When reading the documentation I followed the first 3 snippets to get started, but nothing gets picked up with ```device.name``` I can read ```device.id``` but I don't know which device it's giving me because the device id that is being read doesn't match the addresses I know of. So when I tried to read the device object it just shows an object full of errors, but at the beginning, it says "Unknown error occurred. This is probable a bug! Check reason property." I also tried to find the device by the localName, and id properties with no success.

Here is the console.log(device)
```
{"_manager": {"_activePromises": {"4": [Function anonymous]}, "_activeSubscriptions": {"1": [Object]}, 
"_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", 
"103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", 
"404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for 
device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_subscriber": [EventSubscriptionVendor]}, "_scanEventSubscription": {"context": 
undefined, "emitter": [NativeEventEmitter], "eventType": "ScanEvent", "key": 0, "listener": [Function scanListener], "subscriber": [EventSubscriptionVendor]}, "_uniqueId": 4}, "id": "5E:5A:AD:6C:F4:51", "isConnectable": null, "localName": null, "manufacturerData": "BgABCSAC7K2d0AohW7jonrTvS8EFFFwol438xPQ=", "mtu": 23, "name": null, "overflowServiceUUIDs": null, "rssi": -55, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}
```

And this is my current code related to this module
```
import {BleManager} from 'react-native-ble-plx';

export default class App extends React.Component {
    constructor(props){
        super(props)
        this.manager = new BleManager();
    }

    scanAndConnect = () => {        
        this.manager.startDeviceScan(null, null, (error, device) => {
            if ( error) {                
                console.error(error)
                return error
            }
            
            if ( device ) {
                console.log(device, " found!")
                this.manager.stopDeviceScan();
            }            
        })
    }
    componentDidMount() {
        const subscription = this.manager.onStateChange((state) => {
            if ( state === 'PoweredOn' ) {
                this.scanAndConnect()
                subscription.remove()
            }
        }, true)
    }
```

My manifest permissions are
```
<uses-permission android:name="android.permission.INTERNET" />
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```

One thing that might be good to note is that this project is for the purpose of me learning to work with Bluetooth devices, so I have no previous experience with Bluetooth or BLE at all. And to be honest I don't even know if my phone or the devices I am trying to connect to support BLE. Any suggestions appreciated. If I posted this in the wrong location, I apologize, I just saw other questions posted here. Thanks in advance.

Device OS: Android
react-native: 0.63.0
react-native-ble-plx: ^2.0.2
react-native-bluetooth-classic: ^1.60.0-rc.7

jsriganesh, 2021-02-06:
Same Problem

closed 2021-06-11 by stale[bot]

ELBEQQAL94, 2021-11-17:
Same problem, This library not working anymore

taj567, 2021-11-26:
same problem it is fetching only icon with bluetooth only not all devices .

farlandliu, 2022-02-15:
```
TypeError: Cannot read properties of undefined (reading 'createClient')

This error is located at:
    in App (created by ExpoRoot)
    in ExpoRoot
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
at node_modules/expo/build/logs/LogSerialization.js:156:14 in _captureConsoleStackTrace
at node_modules/expo/build/logs/LogSerialization.js:41:26 in serializeLogDataAsync
- ... 9 more stack frames from framework internals
```

kacper-ragankiewicz, 2023-12-05:
same here:
```
{"_manager": {"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {}, "_scanEventSubscription": null, "_uniqueId": 3}, "id": "1C:CC:16:0B:5A:E8", "isConnectable": null, "localName": null, "manufacturerData": null, "mtu": 23, "name": "Test advertising raw data", "overflowServiceUUIDs": null, "rssi": null, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}
```
Testing connection on Andorid to Arduino Nano 33 BLE wit ArduinoBLE.h examples

Code on react native: 
```
    const connectToDevice = async (device: Device) => {
      try {
        console.log(await device) // This print upper object

        const deviceConnection = await bleManager.connectToDevice(device.id);
        await setConnectedDevice(deviceConnection);
        await deviceConnection.discoverAllServicesAndCharacteristics();
        bleManager.stopDeviceScan();

        startStreamingData(deviceConnection);

      } catch (e) {
        console.log("FAILED TO CONNECT", e);
      }
    };
```