dotintent/react-native-ble-plx #809: onDisconnected never triggered

nicolas-wl, 2021-01-26:
When turning off Bluetooth or physically disconnecting my device "onDisconnected" is never fired 

Platform: iOS
Version: 2.0.2

Simplified code:
```
const connect = async(deviceID) => {
let connectedDev = await manager.connectToDevice(deviceID);
if (connectedDev) {
       connectedDev.onDisconnected((error) => {
            console.log(error);
       });
}
}
```

ThomasStubbe, 2021-01-26:
Having similar issues on Android with Version 2.0.2: If the device is disconnected and turned on quickly afterwards, the device is disconnected, but the onDisconnected event is never thrown.

nicolas-wl, 2021-01-26:
Edit: Issue solved, wasn't a bug. Just bad storing of the BLEManager instance

closed 2021-01-26 by nicolas-wl