dotintent/react-native-ble-plx #813: get advertising data

chocky335, 2021-02-02:
How we can get advertising data? It's required to identify the manufacturer of the device. I've tried to use `manufacturerData ` but didn't get any good progress...

Gueoff, 2021-02-03:
Hello! I just installed the lib and found that:
https://github.com/Polidea/react-native-ble-plx/issues/781

You must decode the "device.manufacturerData" field like this:

```
import RNFetchBlob from 'rn-fetch-blob'; // You can also take another lib like react-native-base64
import {stringToBytes} from 'convert-string';

const base64 = RNFetchBlob.base64;

const advertisingData = stringToBytes(base64.decode(device.manufacturerData));
```

Madhu02, 2021-02-09:
Use this piece of code
import { Buffer } from 'buffer';

const base64ToHex = (str: any) => {
  const decoded = Buffer.from(str, 'base64').toString('hex');
  return decoded;
};

    if (device?.manufacturerData != null) {
      let decoded = classname.base64ToHex(device?.manufacturerData);
      console.log('DECODED:', decoded);
    }

closed 2021-06-11 by stale[bot]