dotintent/react-native-ble-plx #815: BleError: Invalid UUIDs or IDs were passed: null, null,

ayberkgerey, 2021-02-06:
Hello, I can connect to my bluetooth device, When I try to read characteristics I am getting this "BleError: Invalid UUIDs or IDs were passed: null, null," I need to send a string to my device here's my code. please help me.

  const tempDevice = useContext(DeviceContext);
  const manager = new BleManager();

  useEffect(() => {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(JSON.stringify(error));
        return;
      }
      if (device.name === tempDevice.tempCode) {
        console.log('DEVICE CONNECTED.');

        manager.stopDeviceScan();
        // Proceed with connection.
        device
          .connect()
          .then((device) => {
            console.log(
              JSON.stringify(device.discoverAllServicesAndCharacteristics()),
            );
            return device.discoverAllServicesAndCharacteristics();
          })
          .then(async (device) => {
            console.log(
              'device id : ' + device.id + 'device name : ' + device.name,
            );
            console.log(JSON.stringify(device.readCharacteristicForService()));
            console.log('Continue with characteristics and services');
            // Do work on device with services and characteristics
          })
          .catch((error) => {
            // Handle errors
          });
      }
    });
  });

ayberkgerey, 2021-02-09:
I fixed.

closed 2021-02-09 by ayberkgerey

dmozar, 2022-03-18:
How?

theyanniss23002, 2025-04-09:
@ayberkgerey how?????

ayberkgerey, 2025-04-09:
> @ayberkgerey how?????

Bruh, it was 4 years ago. I don't know...