dotintent/react-native-ble-plx #816: Bug Report Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.    > Could not find com.github.Polidea:MultiPlatformBleAdapter:0.1.7.

shuidaan, 2021-02-09:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

    "react": "17.0.1",
    "react-native": "0.63.4",
    "react-native-ble-plx": "^2.0.1",
    "react-native-device-info": "^8.0.1",
    "react-native-gesture-handler": "^1.10.0",
## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

l>react-native run-android
info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.
Jetifier found 1129 file(s) to forward-jetify. Using 4 workers...
info JS server already running.
info Installing the app...

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/4.10.1/userguide/command_line_interface.html#sec:command_line_warnings

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.Polidea:MultiPlatformBleAdapter:0.1.7.
     Searched in the following locations:
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/google/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/google/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/android/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - file:/D:/Users/Administrator.USER-20191030HC/AppData/Local/Android/Sdk/extras/android/m2repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - http://maven.aliyun.com/nexus/content/groups/public/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - http://maven.aliyun.com/nexus/content/groups/public/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - file:/C:/Users/Administrator.USER-20191030HC/.m2/repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - file:/C:/Users/Administrator.USER-20191030HC/.m2/repository/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - https://dl.google.com/dl/android/maven2/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - https://dl.google.com/dl/android/maven2/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - https://jcenter.bintray.com/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - https://jcenter.bintray.com/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
       - file:/D:/rn/react-native-ble-demo/ble_central/node_modules/react-native/android/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.pom
       - file:/D:/rn/react-native-ble-demo/ble_central/node_modules/react-native/android/com/github/Polidea/MultiPlatformBleAdapter/0.1.7/MultiPlatformBleAdapter-0.1.7.jar
     Required by:
         project :app
         project :app > project :react-native-ble-plx

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 9s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup. Run CLI with --verbose flag for more details.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
     "react": "17.0.1",
    "react-native": "0.63.4",
    "react-native-ble-plx": "^2.0.1",
    "react-native-device-info": "^8.0.1",
    "react-native-gesture-handler": "^1.10.0",
* Platform: Android 
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

shuidaan, 2021-02-09:
[image]
need  build.gradle add maven { url 'https://www.jitpack.io' }

tommy-mor, 2021-05-12:
I have same problem

JairajJangleSLIOT, 2022-05-12:
Following this fixed my issue: https://stackoverflow.com/a/72213687/7040601

thegreatzeus, 2022-10-25:
I have the same issue
[image]

gusttavodev, 2022-11-06:
in android/build.gradle file Add follow lines

Top of file
```
def REACT_NATIVE_VERSION = new File(['node', '--print',"JSON.parse(require('fs').readFileSync(require.resolve('react-native/package.json'), 'utf-8')).version"].execute(null, rootDir).text.trim())
```
allprojects item
```
allprojects {
     configurations.all {
        resolutionStrategy {
            force "com.facebook.react:react-native:" + REACT_NATIVE_VERSION
        }
    }
    repositories {
        mavenLocal()
        // ...
}
```