dotintent/react-native-ble-plx #820: I can't write a new value to device.

ayberkgerey, 2021-02-13:
I can connect to device and I can write for the first. When I try to send a second value my device is getting disconnected. I want to send a value outside of the useEffect with a Switch.How can I solve this with this ble library ?
My Code: 

  const scanDevice = () => {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(JSON.stringify(error));
        navigation.navigate('MainScreen', {screen: 'MainScreen'});
        return;
      }
      if (device.name === tempDevice.tempCode) {
        console.log('DEVICE CONNECTED.');

        manager.stopDeviceScan();
        // Proceed with connection.
        device
          .connect()
          .then((device) => {
            console.log('Connected...Discovering services and characteristics');
            return device.discoverAllServicesAndCharacteristics();
          })
          .then(async (device) => {
            console.log('Services and characteristics discovered');
            ToastAndroid.show('Connected', ToastAndroid.SHORT);
            ref.current = device.id;
            console.log('tempID : ');
            console.log(ref);
            // Do work on device with services and characteristics
          })
          .catch(async (error) => {
            console.log(error);
            alert(error.message);
            navigation.navigate('MainScreen', {screen: 'MainScreen'});
            // Handle errors
          });
      }
    });
  };

  useEffect(() => {
    scanDevice();
  });

  const toggleSwitch = async () => {
    setIsEnabled((previousState) => !previousState);
    if (!isEnabled) {
      await manager
        .writeCharacteristicWithResponseForDevice(
          ref.current,
          serviceUUID,
          characteristicUUID,
          AUTOMOD,
          '1',
        )
        .then((value) => console.log(value))
        .catch((err) => console.log(err));
      console.log('ref : ');
      console.log(ref.current);
      console.log('AUTOMOD SENT');
    } else {
      await manager
        .writeCharacteristicWithResponseForDevice(
          ref.current,
          serviceUUID,
          characteristicUUID,
          AUTOMOD,
          '2',
        )
        .then((value) => console.log(value))
        .catch((err) => console.log(err));
      console.log('OFFMOD SENT!');
    }
  };

sweebee, 2021-02-24:
Looks similar to my issue #821. Are you using android? And which android version/phone?

ayberkgerey, 2021-02-24:
@sweebee Yes I am using Android, Version 10 QKQ1.200114.002 /Xiaomi Mi Note 8. Thanks for your respond.

ayberkgerey, 2021-02-26:
@sweebee Do you have any idea about it's solution?

sweebee, 2021-02-26:
Unfortunately not. I don't have a psychical device with this problem, only logs from Bugsnag. I'm still looking for a device with this problem, tested like 6 android phones from colleagues and they all worked fine.

closed 2021-06-02 by ayberkgerey