dotintent/react-native-ble-plx #821: Some devices get write fails (Android)

sweebee, 2021-02-22:
We are using this package in an app with around 300 users. These users have to setup a BLE device. But some (only android) users (10?) just keep getting this error (logged with Bugsnag):

```
BleError: Characteristic 00004e03-0000-1000-8000-00805f9b34fb write failed for device 84:DF:0C:10:2C:AB and service 00004e01-0000-1000-8000-00805f9b34fb
```

I'm unable to track this with any local devices we have, but what could this be? Is it an issue with this package? The BLE device? or a specific android version?

ayberkgerey, 2021-02-24:
@sweebee Did you check characteristics ? Try this if you did not.

   const services = await device.services();
            const characteristics0 = await services[0].characteristics();
            const characteristics1 = await services[1].characteristics();
            const characteristics2 = await services[2].characteristics();
            console.log('Characteristics0: ', characteristics0);
            console.log('Characteristics1: ', characteristics1);
            console.log('Characteristics2: ', characteristics2);

sweebee, 2021-02-24:
@ayberkgerey yes this way:

```
    await this.connectedDevice.discoverAllServicesAndCharacteristics();
    const services = await this.connectedDevice.services();
    this.characteristics = [];
    for await (const service of services) {
      const characteristics = await service.characteristics();
      for await (const characteristic of characteristics) {
        this.characteristics.push(characteristic)
      }
    }
```

sweebee, 2021-02-24:
I received device info from 2 customers that are using the app, they were both using a Samsung Galaxy S10 with Android 11.

ayberkgerey, 2021-02-25:
@sweebee My project has to work over anroid 5.

Lacni135, 2021-04-22:
I am having the same issue but only seems to happen when i am plugged with a usb cable to a comptuer with file transfer unabled?? 
Using a Oneplus 7 Pro on Android 10

tlechman49, 2021-08-20:
I'm seeing a similar error on Galaxy Note 10. Has anyone found a solution?

JairajJangleSLIOT, 2022-08-17:
Found a workaround to this solution: 
Add a slight delay before writing to the BLE device, I used a 50ms delay before writing anything to the BLE device to solve this issue.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska