dotintent/react-native-ble-plx #822: Polidea x Snowflake acquisition

StephaneJuban, 2021-02-25:
I am wondering is the recent acquisition of Snowflake on Polidea will change anything about this library ? Will it be still supported and/or improved ?

Thanks for the feedback !

StephaneJuban, 2021-05-20:
_Feeling worried..._

dariuszseweryn, 2021-05-22:
Corporate world needs corporate agreements. This takes time unfortunately. I (or someone else) will follow up once there will be some conclusions. I can assure you that the topic is still rolling. ;)

fabiendem, 2021-09-24:
Hello,

Following Polidea/Snowflake acquisition, it looks like `dotintent` took over the repo.
- Is dotintent the new name of Polidea?
- Will updates be published to the existing npm repo? https://www.npmjs.com/package/react-native-ble-plx
- I see active discussions on PRs so I am assuming maintenance is still active, can you confirm?

I am trying to get a sense of the health of the library.

Thanks a lot for the amazing work so far.

Cheers

StephaneJuban, 2021-10-31:
Hi !

Maybe @bor-ski  or @dariuszseweryn can give us few insights ?

Thanks ! 👍

mciechanowicz, 2022-01-04:
Hi! ✋

answering the questions:
- no, dotintent is not a new name for Polidea. It's a different company and now dotintent will be dealing with the library,
- the dotintent team will be developing the library and fixing existing bugs 😉

You can continue to use the library in your projects without fear of it being abandoned 🚀

closed 2022-01-04 by mciechanowicz

zabojad, 2022-12-29:
We have no new commit since Dec 9th 2021 and pull requests piling up... Should we still be confident with the future of this lib in 2023 ?

fabiendem, 2023-01-05:
@zabojad I would consider alternatives such as https://github.com/innoveit/react-native-ble-manager
See thread at https://github.com/dotintent/react-native-ble-plx/issues/908#issuecomment-1214912359