dotintent/react-native-ble-plx #824: [Android] destroying the manager the app crashes returning destroyClient on a null object reference

Fannolo, 2021-03-01:
## Expected Behavior

When I destroy the BleManager I hope it doesn't destroy

## Current Behavior

When the BleManager is destroyed the app on Android  crashes returning destroyClient on a null object reference

### Steps to Reproduce

1. create a new instance of BleManager on an Android device
2. subsequently call destroy on a componentWillUnmount

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.2
* Platform: Android.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file: 
```js 
{ 
  "dependencies": {
    "@react-native-community/async-storage": "^1.12.1",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/netinfo": "^5.9.9",
    "@react-navigation/bottom-tabs": "^5.8.0",
    "@react-navigation/native": "^5.7.3",
    "@react-navigation/stack": "^5.9.0",
    "axios": "^0.21.1",
    "i18n-js": "^3.7.1",
    "react": "16.13.1",
    "react-native": "0.63.4",
    "react-native-base64": "^0.1.0",
    "react-native-ble-plx": "^2.0.1",
    "react-native-permissions": "^3.0.0",
    "react-native-pixel-perfect": "^1.0.2",
    "react-native-vector-icons": "^7.1.0",
    "react-redux": "^7.2.1",
    "redux": "^4.0.5",
    "redux-thunk": "^2.3.0",
    "uuid-int": "^3.0.3"
  },
  "devDependencies": {
    "@babel/core": "^7.8.4",
    "@babel/runtime": "^7.8.4",
    "@react-native-community/eslint-config": "^1.1.0",
    "babel-jest": "^25.1.0",
    "eslint": "^6.5.1",
    "jest": "^25.1.0",
    "metro-react-native-babel-preset": "^0.59.0",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  }
}
```
* Formatted code sample or link to a repository:

```js
 destroy() {
    this.setState("scanning", false);
    this.manager.destroy()
  }
```

Fannolo, 2021-03-01:
I solved it by adding a control in BleClientManager.java
```java
@ReactMethod
    public void destroyClient() {
        if(bleAdapter != null){
            bleAdapter.destroyClient();
        }
        bleAdapter = null;
    }
```

closed 2021-06-11 by stale[bot]

oskarhertzman, 2022-06-07:
Would recommend Fannolo's fix in combination with [Patch package](https://www.npmjs.com/package/patch-package)