dotintent/react-native-ble-plx #827: Bug Report

Alexispap, 2021-03-02:
## Prerequisites

- [x ] I am running the latest version
- [x ] I checked the documentation and found no answer
- [x ] I checked to make sure that this issue has not already been filed

## Expected Behavior

An existing characteristic is not found with iOS, though it appears on Android devices 

## Current Behavior

Search for characteristics on a specific service results in empty list.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Connect to device
2. Find service
3. Get the characteristics of the service --> nothing found (but is definitely there)

### Context
* Library version: 2.0.2
* Platform: iOS.

Is there something to take into account that is different for Iphones here?
Trying out another package also does not give the characteristics with iOS.

dariuszseweryn, 2021-04-14:
Yes, iOS strips some characteristics to protect user's location/privacy

closed 2021-04-14 by dariuszseweryn

Alexispap, 2021-04-14:
Are they listed somewhere? I wonder which will work.

dariuszseweryn, 2021-04-14:
I couldn't find anything specific for 15 minutes. I just remember that iOS did not allow for access to everything that is in Bluetooth Spec with the CoreBluetooth library.