dotintent/react-native-ble-plx #833: [IOS] Release build fails with linking errors to swift libs despite bridging header

ortonomy, 2021-03-17:
## Expected Behavior

App builds for release

## Current Behavior

App fails on release build with errors relating to swift libraries used in MultiPlatformBLEAdapater

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. install `react-native-ble-plx` as per README
2. build for release

### Context

[image]

[image]

[image]

* Library version: 2.0.2
* Platform: iOS.
* Platform logs (logcat/XCode):

```

Showing Recent Issues
Ld /Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Intermediates.noindex/onigiri.build/Release-iphonesimulator/onigiri.build/Objects-normal/arm64/Binary/onigiri normal arm64 (in target 'onigiri' from project 'onigiri')
    cd /Users/ortonomy/Vault/projects/professional/aliengen/clients/nuggets_technologies/onigiri/code/onigiri/ios
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -target arm64-apple-ios10.0-simulator -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator14.4.sdk -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/iphonesimulator -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift-5.0/iphonesimulator -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/BVLinearGradient -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/DoubleConversion -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/FBReactNativeSpec -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/Folly -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/MultiplatformBleAdapter -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/Permission-LocationWhenInUse -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RCTTypeSafety -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNBootSplash -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNCAsyncStorage -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNCMaskedView -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNGestureHandler -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNLanguages -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNLocalize -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNPermissions -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNReanimated -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNSVG -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/RNScreens -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-Core -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-CoreModules -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTAnimation -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTBlob -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTImage -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTLinking -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTNetwork -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTSettings -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTText -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-RCTVibration -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-cxxreact -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-jsi -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-jsiexecutor -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/React-jsinspector -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/ReactCommon -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/Yoga -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/glog -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/react-native-ble-plx -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/react-native-geolocation -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/react-native-safe-area-context -L/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator/react-native-webview -F/Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Products/Release-iphonesimulator -filelist /Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Intermediates.noindex/onigiri.build/Release-iphonesimulator/onigiri.build/Objects-normal/arm64/onigiri.LinkFileList -Xlinker -rpath -Xlinker /usr/lib/swift -Xlinker -rpath -Xlinker /usr/lib/swift -Xlinker -rpath -Xlinker @executable_path/Frameworks -dead_strip -Xlinker -object_path_lto -Xlinker /Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Intermediates.noindex/onigiri.build/Release-iphonesimulator/onigiri.build/Objects-normal/arm64/onigiri_lto.o -Xlinker -objc_abi_version -Xlinker 2 -fobjc-arc -fobjc-link-runtime -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/iphonesimulator -L/usr/lib/swift -Xlinker -add_ast_path -Xlinker /Users/ortonomy/Library/Developer/Xcode/DerivedData/onigiri-ekuxdnmnnxjdiidqkasyrbzwwqbd/Build/Intermediates.noindex/onigiri.build/Release-iphonesimulator/onigiri.build/Objects-normal/arm64/onigiri.swiftmodule -ObjC -lBVLinearGradient -lDoubleConversion -lFBReactNativeSpec -lFolly -lMultiplatformBleAdapter -lPermission-LocationWhenInUse -lRCTTypeSafety -lRNBootSplash -lRNCAsyncStorage -lRNCMaskedView -lRNGestureHandler -lRNLanguages -lRNLocalize -lRNPermissions -lRNReanimated -lRNSVG -lRNScreens -lReact-Core -lReact-CoreModules -lReact- [... cut ...]
    "@babel/eslint-parser": "^7.12.1",
    "@babel/preset-flow": "^7.10.1",
    "@babel/runtime": "^7.8.4",
    "@react-native-community/eslint-config": "^1.1.0",
    "babel-jest": "^25.1.0",
    "eslint": "^7.18.0",
    "eslint-plugin-flowtype": "^5.2.0",
    "flow-bin": "^0.126.1",
    "jest": "^25.1.0",
    "metro-react-native-babel-preset": "^0.59.0",
    "react-native-dotenv": "^2.4.3",
    "react-native-svg-transformer": "^0.14.3",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

ortonomy, 2021-03-17:
xcode version is 12.4

Brassco, 2021-03-23:
I got this errors for Debug scheme

tickietackie, 2021-05-23:
Just wanted to say that I got the same error, but indeed it was related to not having the bridge file. Somehow the first time I was creating the first swift file I wasn't ask for if a bridging file should be created. So I deleted the Swift file, created it again and the second time it asked me. Afterwards the build succeeded. This probably does not have sth. do to with this issues in particular, but maybe someone stumbles over the same problem that I did.

tickietackie, 2021-06-20:
Also not opening the workspace file directly got me this error.