dotintent/react-native-ble-plx #835: Trying to get heartbeat out of BLE Heartbeat monitor with React Native

fre887, 2021-03-24:
Hi friends,

Hope you can help me, I'm a bit stuck right now ...
Trying to make a React Native IOS app, but can't seem to get the Bluetooth component to work.

Here is my code: 

`manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(error.message);
        return;
      }
      if (device.name === "TICKR FIT FB91") {
        console.log("Connecting to:", device.name);
        manager.stopDeviceScan();
        device.connect().then(() => {
          device.monitorCharacteristicForService(
            "180D",
            "2A37",
            (error, characteristic) => {
              if (error) {
                console.log(error.message);
                return;
              }
              console.log("notified");
              console.log(characteristic.uuid);
              console.log(characteristic.value.toString());
            }
          );
        });
      }
    });`

UUID of the Heartbeat Service is 180D and the heart beat measurements are 2A37. Gettin' errors about a "Possible Unhandled Promise Rejection" and the message that "2A37" isn't a Characteristic.

Can someone please help me to get the heartbeat measurement characteristic out of the Bluetooth wristband with the use of React Native BLE PLX.

Greetz,

Fre

hugows, 2021-03-24:
Hello friend,
I feel like you're missing some steps, like discoverAllServicesAndCharacteristics which is actually required even if you know your characterestic and service IDs - the library needs to know, lets say:

_Discovery of services and characteristics is required to be executed once per connection*. It can be a long process depending on number of characteristics and services available._

Take a look at https://polidea.github.io/react-native-ble-plx/ again and maybe try to run the https://github.com/PolideaPlayground/SensorTag sample project. 
Good luck!

fre887, 2021-03-25:
Hi Hugo,

Thanks a lot! As you suggested, that was indeed the issue. My code works now. I'll post the code for other people who might run in to the same issue as me.

```

manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(error.message);
        return;
      }
      if (device.name === "TICKR FIT FB91") {
        console.log("Connecting to:", device.name);
        manager.stopDeviceScan();
        device.connect()
          .then(() => {
            return device.discoverAllServicesAndCharacteristics();
          })
          .then(() => {
            device.characteristicsForService("180D").then((characteristics) => {
              return characteristics;
            });
            return device.monitorCharacteristicForService();
          })
          .then(() => {
            device.monitorCharacteristicForService(
              "180D",
              "2A37",
              (error, characteristic) => {
                if (error) {
                  console.log(error.message);
                  return;
                }
                const buffer = Buffer(characteristic.value, "base64");
                value = buffer[1];
                return value;
              }
            );
          });
      }
    });

```

closed 2021-03-25 by fre887