dotintent/react-native-ble-plx #849: Current device id

quannguyennn, 2021-05-20:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [ x] I checked the documentation and found no answer
- [x ] I checked to make sure that this issue has not already been filed
- [ x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

I want to build an application to tracking exposure to covid patient.
I was able to scanning available around the device. However, i can't get my own device id.

ezranbayantemur, 2021-06-02:
How you can't get device id? It's storing on `device.id` property

GlobusHugoSchmitt, 2021-06-02:
Also note that device.id is not Mac Address on iOS: "Device identifier: MAC address on Android and UUID on iOS."

ezranbayantemur, 2021-06-02:
Yes you can't see the mac address on iOS but it's not a bug. iOS disable it by default and generates an unique id. 
For seeing mac id the ble device should send it by with manufacturer data or something else. Example our project: our devices sent it's MAC address with manufacturerData property and I parse it to the macId

tmjordan, 2021-06-13:
Hey guys,
Please can you explain me why on an iphone and on android, for the same device we don't have the same device id (whether it be uuid or mac address)

tmjordan, 2021-06-13:
Hey @ezranbayantemur , if iOS generates a random unique id, why can't we get it. I'm looking for a way to associate a device id to an account on my server, so i need to access to a local identifier and i need it to be accessible by other devices using scan.
But scan function, on android returns a mac address for an iOS device, while iOS returns an UUID for an android device.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska