dotintent/react-native-ble-plx #853: Null pointer exception when we unsubscribe the monitor and clear the ble instance

arpitgarg23, 2021-05-26:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

Issue with 2.0.2

## Expected Behavior
It should have null check for bleAdpater

   ```
 
```

### Steps to Reproduce
Terminate and close the ble connection with device.
Please provide detailed steps for reproducing the issue.

### Context

```
2021-05-26 20:37:19.192 27037-27173/com.xx.xx E/unknown:ReactNative: Exception in native call
    java.lang.NullPointerException: Attempt to invoke interface method 'void com.polidea.multiplatformbleadapter.BleAdapter.cancelTransaction(java.lang.String)' on a null object reference
        at com.polidea.reactnativeble.BleClientManager.cancelTransaction(BleClientManager.java:107)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
        at android.os.Looper.loop(Looper.java:164)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
        at java.lang.Thread.run(Thread.java:764)
```

* Library version: X.Y.Z
* Platform: Android/
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:
```

@ReactMethod
    public void destroyClient() {
        bleAdapter.destroyClient();
        bleAdapter = null;
    }

    // Mark: Common --------------------------------------------------------------------------------

    @ReactMethod
    public void cancelTransaction(String transactionId) {
        bleAdapter.cancelTransaction(transactionId);
    }
```

Jurpp, 2022-05-25:
We are experiencing this bug as well. Could you confirm this issue is not present in 2.0.1?

scinfu, 2023-04-06:
bug still present in 2.0.3

dominik-czupryna-withintent, 2023-10-02:
If I understand you correctly, you destroyed the adapter and then called a function on it. You won't be able to call functions from it because it no longer exists.

closed 2023-10-02 by dominik-czupryna-withintent