dotintent/react-native-ble-plx #855: How to pair bluetooth devices programatically without user interaction using react-native-ble-plx?

aarthigovardhanan88, 2021-05-31:
## Question

I am trying to implement a Bluetooth pairing programatically between a device and mobile app. I am able to pair the device but the problem is that it requires user to enter the passcode which I don't want in my app. I want to do that programatically. I need to provide the device name and the passcode programatically. I am using the library react-native-ble-plx. I could not find documentations related to this in the library.

Is this possible with the current version of the library?

ezranbayantemur, 2021-06-02:
What you mean passcode? Should you write it to device characteristic?

aarthigovardhanan88, 2021-06-10:
No, Passcode/password used for pairing

ezranbayantemur, 2021-06-10:
You can create a modal which is contains a TextInput for asking user to enter PIN and then you can let the user pair the device.

And plus, it's not a bug or a issue of this package. I suggest you to ask these kind of code questions to StackOverflow. Not here.

closed 2022-01-04 by mciechanowicz

etruscanarcher, 2024-06-28:
How did you solve this problem? Can you specify it?