dotintent/react-native-ble-plx #857: Support for react-native 0.64.2

williamgoodhew, 2021-06-07:
Is there a timeline for support for react-native 0.64.2?

orther, 2021-06-09:
I was planning to upgrade a project using react-native-ble-plx from `0.64.1` to `0.64.2` soon. Are there problems with 0.64.2 and this library?

ezranbayantemur, 2021-06-10:
I don't think any pre-requests for about this package on 0.64.2 version. Seems like there is not any changes on documentation for native side which is affect this library. 
https://github.com/facebook/react-native/releases/tag/v0.64.2

Plus I already updated my project to 0.64.2 and there is not any issue or problem occured.

Maybe something else? @williamgoodhew

zardazare, 2021-06-15:
> I was planning to upgrade a project using react-native-ble-plx from `0.64.1` to `0.64.2` soon. Are there problems with 0.64.2 and this library?

Build command for testing is failing: 

`./gradlew assembleDebug assembleAndroidTest -DtestBuildType=debug`

Build command is used in a combination with detox build/detox test configurations.

Task :react-native-ble-plx:processDebugAndroidTestManifest FAILED
[androidx.vectordrawable:vectordrawable-animated:1.0.0] $HOME/.gradle/caches/transforms-2/files-2.1/ec3fddf24b26d7deb84606b8aa100c87/vectordrawable-animated-1.0.0/AndroidManifest.xml Warning:
        Package name 'androidx.vectordrawable' used in: androidx.vectordrawable:vectordrawable-animated:1.0.0, androidx.vectordrawable:vectordrawable:1.0.1.
$HOME/projectName/node_modules/react-native-ble-plx/android/build/intermediates/tmp/manifest/androidTest/debug/manifestMerger2812710704197426270.xml:5:5-74 Error:
        uses-sdk:minSdkVersion 18 cannot be smaller than version 21 declared in library [com.facebook.react:react-native:0.64.2] $HOME/.gradle/caches/transforms-2/files-2.1/91d9154e6c9cdfb163556c5c86d67a1e/jetified-react-native-0.64.2/AndroidManifest.xml as the library might be using APIs not available in 18
        Suggestion: use a compatible library with a minSdk of at most 18,
                or increase this project's minSdk version to at least 21,
                or use tools:overrideLibrary="com.facebook.react" to force usage (may lead to runtime failures)

hugows, 2021-07-22:
0.64.1 works?

zorba0517, 2021-09-14:
> Build command for testing is failing:
>
> `./gradlew assembleDebug assembleAndroidTest -DtestBuildType=debug`
>
> Build command is used in a combination with detox build/detox test configurations.
>
> Task [...]

i have same problem

zardazare, 2021-09-22:
> 0.64.1 works?

I did not try to decrease react-native version. In the end, why should I, I have other stuff related to the newer version.

I solved this particular issue with patch (https://www.npmjs.com/package/patch-package), increasing **minSdkVersion 21** @zorba0517.

mciechanowicz, 2021-12-07:
Hi everyone ✋

We already have PR with React(17.0.2) and React Native(0.66.3) version upgrade here: https://github.com/dotintent/react-native-ble-plx/pull/925

I hope it will be merged soon and the new version of the library will be released! 🚀

closed 2022-01-13 by mciechanowicz