dotintent/react-native-ble-plx #858: The manufacturer data is not being updated.

VaruzhKhachatryan, 2021-06-09:
## Prerequisites

 I am running the 2.0.0 version

After the BLE manufacturer data is being updated, I destroy the BLEManager, creating new one and start scanning devices, I can see that the manufacturer data is the previous one (not the updated one).

But when I connect to the device, disconnect and scan the devices again, I can see the updated BLE manufacturer data.

Is there any way to scan the the device with updated manufacturer data, without connecting/disconnecting ?

DaBs, 2021-09-09:
This highly depends on your device - does the device immediately update manufacturer data? Can you verify this with an external app? Some devices only update manufacturer data when needed, e.g. after a disconnection and subsequent re-transmission of advertising and scan response data.

mciechanowicz, 2022-01-13:
Hi. Is @DaBs answer sufficient for you? I'm closing it, for now, reopen this issue if you have any additional questions.

closed 2022-01-13 by mciechanowicz