dotintent/react-native-ble-plx #860: Understanding BleError: Invalid UUIDs or IDs were passed

slarkin64, 2021-06-10:
## Question
I am trying to connect to a proprietary peripheral over ble and I am running into an issue with writing to a particular characteristic. 

The error I receive back is 
[BleError: Invalid UUIDs or IDs were passed: 07000000-E844-4E43-8AB5-8930F76B0FED,  07000010-E844-4E43-8AB5-8930F76B0FED, ]
 
I have looked through the documentation but can't work out what this means exactly.

There are other places within the application that use _writeCharacteristicWithResponseForService_ without issue so I would like to know more about this error as it could simply be a problem with our proprietary peripheral but I need clarity to investigate that further.

Thanks for any help you can provide

ezranbayantemur, 2021-06-10:
So, the error message says the all I think? You sending invalid uuid's to device.

Example from my code: 
```js
enum UUIDS {
  INFO_SERVICE = '0000a234-b38d-4985-720e-0f993a68ee41',
  BATTERY_CHR = '0000a238-b38d-4985-720e-0f993a68ee41',
  ...
  ...
 }

device.readCharacteristicWithResponseForService(
    UUIDS.INFO_SERVICE,
    UUIDS.BATTERY_CHR
  );
```

Basically; If I send "abc" to `INFO_SERVICE` or `BATTERY_CHR` I'll receive the same error as like yours. Because my device has not any service or characteristic like "abc".

That's all.

slarkin64, 2021-06-11:
@ezranbayantemur Thanks for replying. 
Indeed, the error does say all that needs to be said. My confusion was actually due to an error in the characteristic uuid string that I was trying to write to.

closed 2021-06-11 by slarkin64