dotintent/react-native-ble-plx #870: Question : Can I use this library to send data to bluetooth module HM10 which is BLE module?

viren-vii, 2021-06-24:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question
Can I use this library to send data to bluetooth module HM10 which is BLE module?
I want to be able to send some data from my mobile to bluetooth module HM10. Can I send the data from my android phone? I am using react-native to build the application.
Thanking you in advance!

mkirank, 2021-06-24:
Yes, We use this module with a HM-10 (and HM10 clones). Our app is available on both android and ios.

viren-vii, 2021-06-25:
> Yes, We use this module with a HM-10 (and HM10 clones). Our app is available on both android and ios.

Hey thanks for a reply @mkirank ! Can you just hint me on following?
Right now I am not having the module, so for that I am testing it on mobile phone expecting it'll work on module. So is it going to work as I expect?
If yes then I am having problem in writing data! I followed flow like, getting bonded devices and then connecting one of then, and when I try to write on the connected device it gives error `write failed`
If no then can I use service IDs and characteristic IDs from documentation of module and expect it to work properly whenever I'll have module in my hand?

mkirank, 2021-06-25:
Here is a snippet of how I write

` 
// We convert our string to stringtobyte

//convert byte to base64
formatValue = new Buffer(value, 'base64').toString('ascii');
  return new Promise((resolve, reject) => {
      this.manager
        .writeCharacteristicWithResponseForDevice(
          this.peripheralId,
          '0000ffe0-0000-1000-8000-00805f9b34fb',
          '0000ffe0-0000-1000-8000-00805f9b34fb',
          formatValue,
          transactionId,
        )   
        .then(
          characteristic => {
            console.log('write success', value);
            resolve(characteristic);
          },  
          error => {
            console.log('write fail: ', error);

            reject(error);
          },  
`

viren-vii, 2021-06-25:
Does characteristic UUID and service UUID stay same for all modules?

acro5piano, 2021-08-16:
I have the similar problem, and it turns out that Bluetooth version mismatch.

closed 2021-08-30 by viren-vii