dotintent/react-native-ble-plx #871: Bug ReportTask :react-native-ble-plx:compileDebugJavaWithJavac FAILED

xuyushiguang, 2021-06-25:
## Task :react-native-ble-plx:compileDebugJavaWithJavac FAILED

 %$   npx react-native run-android 
info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.
Jetifier found 1101 file(s) to forward-jetify. Using 8 workers...
info JS server already running.
info Installing the app...
Starting a Gradle Daemon, 1 incompatible and 2 stopped Daemons could not be reused, use --status for details

> Task :react-native-ble-plx:compileDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.7/userguide/command_line_interface.html#sec:command_line_warnings
87 actionable tasks: 3 executed, 84 up-to-date
警告: [options] 源值7已过时, 将在未来所有发行版中删除
警告: [options] 目标值7已过时, 将在未来所有发行版中删除
警告: [options] 要隐藏有关已过时选项的警告, 请使用 -Xlint:-options。
3 个警告

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-ble-plx:compileDebugJavaWithJavac'.
> java.lang.IllegalAccessError: class org.gradle.internal.compiler.java.ClassNameCollector (in unnamed module @0x1492330f) cannot access class com.sun.tools.javac.code.Symbol$TypeSymbol (in module jdk.compiler) because module jdk.compiler does not export com.sun.tools.javac.code to unnamed module @0x1492330f

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 14s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
警告: [options] 源值7已过时, 将在未来所有发行版中删除
警告: [options] 目标值7已过时, 将在未来所有发行版中删除
警告: [options] 要隐藏有关已过时选项的警告, 请使用 -Xlint:-options。
3 个警告

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-ble-plx:compileDebugJavaWithJavac'.
> java.lang.IllegalAccessError: class org.gradle.internal.compiler.java.ClassNameCollector (in unnamed module @0x1492330f) cannot access class com.sun.tools.javac.code.Symbol$TypeSymbol (in module jdk.compiler) because module jdk.compiler does not export com.sun.tools.javac.code to unnamed module @0x1492330f

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 14s

info Run CLI with --verbose flag for more details.

xuyushiguang, 2021-06-25:
在安卓手机上出现这个错误, 在安卓模拟器上可正常运行.

fillybrese, 2021-12-14:
I'm experiencing this same issue.

dominik-czupryna-withintent, 2023-10-02:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-10-02 by dominik-czupryna-withintent