dotintent/react-native-ble-plx #881: Temperature data

mustafauyysl, 2021-07-15:
Hello guys. I work temperature sensor with ble. My code is 

`
  useEffect(() => {
    manager.onStateChange(state => {
      const subscription = manager.onStateChange(state => {
        if (state === 'PoweredOn') {
          scanAndConnect();
          subscription.remove();
        }
      }, true);
      return () => subscription.remove();
    });
  }, [manager]);

  const scanAndConnect = () => {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        return;
      }

      if (device.name === 'DEVICE_NAME') {
        connectDevice(device);
        manager.stopDeviceScan();
      }
    });
  };

  const connectDevice = device => {
    device
      .connect()
      .then(device => {
        readData(device);
      })
      .then(device => {
        manager.stopDeviceScan();
      })
      .catch(error => {});
  };

  const readData = async device => {
    await device.discoverAllServicesAndCharacteristics();
    const services = await device.services();
    services.forEach(async service => {
      const characteristics = await device.characteristicsForService(
        service.uuid,
      );
      characteristics.map(x => console.log('Characteristic', x));
    });
  };
`

I see my characteristics. A character is a big object like this

`
      },
      "_eventEmitter":{
         "_disableCallsIntoModule":false,
         "_nativeModule":[
            "Object"
         ],
         "_subscriber":[
            "EventSubscriptionVendor"
         ]
      },
      "_scanEventSubscription":null,
      "_uniqueId":10
   },
   "deviceID":"C4CA674D-140E-516D-0580-C07E7C7523A9",
   "id":10765694560,
   "isIndicatable":false,
   "isNotifiable":false,
   "isNotifying":false,
   "isReadable":true,
   "isWritableWithResponse":true,
   "isWritableWithoutResponse":false,
   "serviceID":10783116672,
   "serviceUUID":"0000fff0-0000-1000-8000-00805f9b34fb",
   "uuid":"0000fff6-0000-1000-8000-00805f9b34fb",
   "value":"AA=="
`

But ı don't see temperature value in this object. How can ı get temperature data from my character ?

hugows, 2021-07-16:
You still have to read the characteristic: https://github.com/dotintent/react-native-ble-plx/wiki/Characteristic-Reading

mustafauyysl, 2021-07-17:
> You still have to read the characteristic: https://github.com/dotintent/react-native-ble-plx/wiki/Characteristic-Reading

But still ı cannot get temperature value

mustafauyysl, 2021-07-17:
> You still have to read the characteristic: https://github.com/dotintent/react-native-ble-plx/wiki/Characteristic-Reading

Thank you so much. I can read my characteristic value. But it is base64 (Dy4=). How can ı convert Celsius ?

closed 2021-07-26 by mustafauyysl

hugows, 2021-07-26:
I think I use [this](https://www.npmjs.com/package/buffer):

`Buffer.from(b64string, 'base64'); // Ta-da`

GiuseppeEpifani, 2021-09-01:
I leave you the formula to decipher the data of a thermometer

https://gist.github.com/GiuseppeEpifani/b994b49c049d585cf3b3dc35ed7e9898