dotintent/react-native-ble-plx #882: Link to documentation in README.md is broken

Jurpp, 2021-07-16:
## Expected Behavior

Link in the README points towards : https://dotintent.github.io/react-native-ble-plx/

## Current Behavior

Link in the README points towards : https://polidea.github.io/react-native-ble-plx/

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Go to the [documentation section in the README](https://github.com/dotintent/react-native-ble-plx#documentation--support).
2. Click on "Documentation can be found here."
3. Observe 404

Jurpp, 2021-07-16:
This is actually a duplicate of #867

mciechanowicz, 2021-12-07:
Hi,

This is actual link to documentation: [documentation](https://dotintent.github.io/react-native-ble-plx/)

closed 2021-12-07 by mciechanowicz