dotintent/react-native-ble-plx #888: Compile fails in Swift 5.5 and Xcode beta

bostnis, 2021-07-27:
I am running the latest version - 2.0.2
I've installed xcode 13 beta and tried to compile the project which includes react-native-ble-plx@^2.0.2 and the build constantly fails due to MultiplatformBleAdapter Pods dependency. 

I could see there was already PR in this link - https://github.com/dotintent/MultiPlatformBleAdapter/pull/80

It would be great if you can merge this PR and publish new version. 
Warm Regards

aditya23jaitly, 2021-09-21:
Can you tell any other way possible to run without these changes in ios 15 device as application is not able to run in ios 15 or can you tell when new version would build with these changes.

JaapWeijland, 2021-09-21:
Does somebody have a patch-package in place? I can't get it to run.

arelstone, 2021-09-22:
I'd really love to see this fixed, either by a patch-package or a new version released

Tadejo9, 2021-09-22:
[Source](https://github.com/dotintent/MultiPlatformBleAdapter/issues/79#issuecomment-924643695)

For the time being, issue can be fixed by making the following changes to your Pods.

**Pods/MultiplatformBleAdapter/RestoredState.swift**
line: 68
```
peripheral: RxCBPeripheral(peripheral: $0.service.peripheral))
```
to
```
peripheral: RxCBPeripheral(peripheral: $0.service.peripheral!))
```
**Pods/MultiplatformBleAdapter/RxCBCharacteristic.swift**
line: 59
```
return RxCBService(service: characteristic.service)
```
to
```
return RxCBService(service: characteristic.service!)
```
**Pods/MultiplatformBleAdapter/RcCBDescriptor**
line: 43
```
return RxCBCharacteristic(characteristic: descriptor.characteristic)
```
to
```
return RxCBCharacteristic(characteristic: descriptor.characteristic!)
```

ball-hayden, 2021-09-22:
If you prefer doing this by `patch-package`, you need two changes. Firstly, the following patch to be made by `patch-packages`:

```patch
diff --git a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
index def210f..91c6eca 100644
--- a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
+++ b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
@@ -17,5 +17,5 @@ Pod::Spec.new do |s|
   s.compiler_flags = '-DMULTIPLATFORM_BLE_ADAPTER'
 
   s.dependency 'React'
-  s.dependency 'MultiplatformBleAdapter', '0.1.7'
+  s.dependency 'MultiplatformBleAdapter', '0.1.8'
 end
```

Then change your `Podfile` (once) as follows:

```patch
+  # Force update to MultiplatformBleAdapter for iOS 15 SDK
+  pod 'MultiplatformBleAdapter', git: "https://github.com/dotintent/MultiPlatformBleAdapter", commit: "669e438acb384eee74f3f1ada3ba259cf7b7965b"
```

Obviously, usual caveats around using unreleased software apply. Very much a hack - thanks Apple for the breaking changes.

arelstone, 2021-09-22:
> If you prefer doing this by `patch-package`, you need two changes. Firstly, the following patch to be made by `patch-packages`:
> 
> ```diff
> diff --git a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> index def210f..91c6eca 100644
> --- a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> +++ b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> @@ -17,5 +17,5 @@ Pod::Spec.new do |s|
>    s.compiler_flags = '-DMULTIPLATFORM_BLE_ADAPTER'
>  
>    s.dependency 'React'
> -  s.dependency 'MultiplatformBleAdapter', '0.1.7'
> +  s.dependency 'MultiplatformBleAdapter', '0.1.8'
>  end
> ```
> 
> Then change your `Podfile` (once) as follows:
> 
> ```diff
> +  # Force update to MultiplatformBleAdapter for iOS 15 SDK
> +  pod 'MultiplatformBleAdapter', git: "https://github.com/dotintent/MultiPlatformBleAdapter", commit: "669e438acb384eee74f3f1ada3ba259cf7b7965b"
> ```
> 
> Obviously, usual caveats around using unreleased software apply. Very much a hack - thanks Apple for the breaking changes.

Will this be compatible with older versions of xCode too?

zardazare, 2021-09-22:
MultiplatformBleAdapter has one merge request which should increase version of this dependency to 0.1.9 https://github.com/dotintent/MultiPlatformBleAdapter/pull/84

This is the issue with iOS15 CoreBluetooth https://github.com/dotintent/MultiPlatformBleAdapter/pull/80

ball-hayden, 2021-09-22:
@JaapWeijland indeed. You need to also update your `Podfile` as I directed. A `pod update MultiplatformBleAdapter` should then behave.

@arelstone no idea.

arelstone, 2021-09-22:
@ball-hayden Seems to be working with xCode 12.5.1 👏  Thanks a lot

JaapWeijland, 2021-09-22:
@ball-hayden I did exactly that, changing both the podspec and the Podfile. This is my patch-package:

```
diff --git a/node_modules/react-native-ble-plx/ios/Podfile b/node_modules/react-native-ble-plx/ios/Podfile
index b6d8923..982a6e9 100644
--- a/node_modules/react-native-ble-plx/ios/Podfile
+++ b/node_modules/react-native-ble-plx/ios/Podfile
@@ -6,5 +6,8 @@ target 'BleClient' do
   use_frameworks!
 
   # Pods for BleClient
-  pod 'MultiplatformBleAdapter', '~> 0.1.7'
+  # pod 'MultiplatformBleAdapter', '~> 0.1.7'
+  # Force update to MultiplatformBleAdapter for iOS 15 SDK
+  pod 'MultiplatformBleAdapter', git: "https://github.com/dotintent/MultiPlatformBleAdapter", commit: "669e438acb384eee74f3f1ada3ba259cf7b7965b"
+
 end
diff --git a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
index def210f..dc81e8b 100644
--- a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
+++ b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
@@ -17,5 +17,5 @@ Pod::Spec.new do |s|
   s.compiler_flags = '-DMULTIPLATFORM_BLE_ADAPTER'
 
   s.dependency 'React'
-  s.dependency 'MultiplatformBleAdapter', '0.1.7'
+  s.dependency 'MultiplatformBleAdapter', '0.1.8'
 end
\ No newline at end of file
```

Running `pod install` and `pod update MultiplatformBleAdapter` does not seem to work even then. Am I overlooking something?

arelstone, 2021-09-22:
> @ball-hayden I did exactly that, changing both the podspec and the Podfile. This is my patch-package:
> 
> ```
> diff --git a/node_modules/react-native-ble-plx/ios/Podfile b/node_modules/react-native-ble-plx/ios/Podfile
> index b6d8923..982a6e9 100644
> --- a/node_modules/react-native-ble-plx/ios/Podfile
> +++ b/node_modules/react-native-ble-plx/ios/Podfile
> @@ -6,5 +6,8 @@ target 'BleClient' do
>    use_frameworks!
>  
>    # Pods for BleClient
> -  pod 'MultiplatformBleAdapter', '~> 0.1.7'
> +  # pod 'MultiplatformBleAdapter', '~> 0.1.7'
> +  # Force update to MultiplatformBleAdapter for iOS 15 SDK
> +  pod 'MultiplatformBleAdapter', git: "https://github.com/dotintent/MultiPlatformBleAdapter", commit: "669e438acb384eee74f3f1ada3ba259cf7b7965b"
> +
>  end
> diff --git a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> index def210f..dc81e8b 100644
> --- a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> +++ b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
> @@ -17,5 +17,5 @@ Pod::Spec.new do |s|
>    s.compiler_flags = '-DMULTIPLATFORM_BLE_ADAPTER'
>  
>    s.dependency 'React'
> -  s.dependency 'MultiplatformBleAdapter', '0.1.7'
> +  s.dependency 'MultiplatformBleAdapter', '0.1.8'
>  end
> \ No newline at end of file
> ```
> 
> Running `pod install` and `pod update MultiplatformBleAdapter` in `./ios` does not seem to work even then. Am I overlooking something?

You should make the changes your own Podfile. It's located in the ios folder.

What I did to make it work:
- Add `pod 'MultiplatformBleAdapter', git: "https://github.com/dotintent/MultiPlatformBleAdapter", commit: "669e438acb384eee74f3f1ada3ba259cf7b7965b"` to my Podfile
- Add the patch file in `./patches`
- ran `yarn patch-package`
- Profit

JaapWeijland, 2021-09-22:
Works! Thanks alot for the help!

mattahorton, 2021-09-22:
So as I understand it, we're just waiting on dotintent/MultiPlatformBleAdapter to make a release of their CoreBluetooth update before we can make a PR here?

JaapWeijland, 2021-09-22:
AFAIK that is correct

mciechanowicz, 2021-12-07:
Hi ✋

This issue is already fixed in 2.0.3 💪 so I am gonna close this ticket