dotintent/react-native-ble-plx #892: BLE Scan not finding devices on some phones

stevenixxdc, 2021-08-17:
So we run the scan function and it works great on my Samsung S21 ultra every time - picks up all the ble devices around and the one I want, fine.  Now the client runs it on an older phone (Alcatel 1 5033x android) and it worked once but after that never worked again.

Alcatel 1 5033x Android Phone

```
BLEManager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        //handle errors
        console.log(error);
        return
      }

      if (device.name != null) {
         ... etc adds device to a list
```

doctorbatch, 2021-09-20:
I have the same issue... except different models. I have a variety of test devices, and my app works perfectly on everything including my Samsung A42 -- however it performs VERY poorly on my Samsung A10E. Advertising packets do not appear to be "seen" and there are very long delays between advertising notifications (measured in minutes) coming from the same device, which is advertising on a 20ms interval.

I have the same problem scanning for my devices using nRF Connect on Android (I do not know what underlying library they use)... so since it is reproducible outside of my react-native-ble-plx application, I have been working under the assumption that there's an underlying issue with the older devices.

Have not found a solution yet -- but glad to see I'm not alone.

t3db0t, 2021-11-14:
I also have discovered that my iPhone 11 works perfectly while a friend's iPhone 12 doesn't detect the expected devices.

dominik-czupryna-withintent, 2023-10-02:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-10-02 by dominik-czupryna-withintent