dotintent/react-native-ble-plx #893: Bug Report

PoieoDev, 2021-08-23:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

We're developing an app for a client which will be required to connect to a heart-rate bluetooth device. What we have is code to determine if bluetooth is in "PoweredOn" state in ios, and if so, scan for devices. If it finds devices capable of serving heart rate data through a specific service, then list them for the user. When the user taps on one of them, the app should connect to that device and start streaming heart rate data to our api. 

## Current Behavior

Everything works great on Android. 
However, on iOS, I continue having issues with the device.connect() method. It will SOMETIMES connect to the device when the app is running through XCode, but as soon as I release it to TestFlight, it NEVER works. And sometimes it doesn't work through XCode either. 

I currently have the following logs:

2021-08-23 13:53:31.413995-0700 mobilefrontend[23420:5608921] [javascript] Scanning
2021-08-23 13:53:31.425763-0700 mobilefrontend[23420:5609561] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x281bf4680> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2021-08-23 13:53:31.427850-0700 mobilefrontend[23420:5609164] [CoreBluetooth] XPC connection invalid

I also get the error "Device is in an Unknown State" (I don't remember the exact wording of it) sometimes when trying to connect. 

Here is my code for this (yes, it's a bit messy. This is a rapid-dev prototype that we just need to get working.)
bleManager is currently set up in a REF hook (bluetoothManager = useRef(bleManager))

```
  const requestBLEPermission = async () => {
  try {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
      {
        title: "Gladys",
        message:
          "Gladys Needs Location Access " +
          "in order to capture Bluetooth Data",
        buttonNeutral: "Ask Me Later",
        buttonNegative: "Cancel",
        buttonPositive: "OK"
      }
    );
    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      scanAndConnect();
    }
  } catch (err) {
  }
};

useEffect(() => {
  if (Platform.OS == "ios") {
    bluetoothManager.current.onStateChange((state) => {
      const subscription = bluetoothManager.current.onStateChange((state) => {
          if (state === 'PoweredOn') {
              startScan();
              subscription.remove();
            }
          }, true);
      return () => subscription.remove();
    });
  } else {
    startScan()
  }
}, [bluetoothManager]);

function startScan() {
  if (allowScan === true) {
  if (Platform.OS === "ios") {
    scanAndConnect()
[... 238 lines cut ...]
            tempList.push(device.name)

            setBtDevicesDictCore(tempDict);
            setBtDevices(tempList)

            addDevices();
          }
    } else if (connectBtID && btDevicesDictCore && btDevices) {
      if (btDevicesDictCore[btDevices[connectBtID]].id === device.id){
      bluetoothManager.current.stopDeviceScan()
      connectBt(connectBtID)
    }}
  } else if (device === null) {
      if(btStatusText !== "Bluetooth not enabled") {
        setBtStatusText("Bluetooth not enabled")
      }
    }
  }});
}
```

This is only a problem in iOS, Android works PERFECT with the existing code.
The problem is exacerbated by pushing to the iOS store/TestFlight.

Any help would be appreciated.

loewj-reekon, 2022-07-19:
I am also facing a similar issue on iOS (with an example similar to yours working on Android). I believe the bug arises from storing the BleManager instance in a ref. Moving it elsewhere lets me invoke the device scan on iOS, but doing so causes some other strange behavior, such as not being able to stop the device scan. I will let you know if I find a workaround.

nepinney, 2022-09-14:
I am experiencing the same issues on IOS. 'BluetoothLE is in unknown state' and in xcode console:
```
[CoreBluetooth] XPC connection invalid
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x280955680> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
``` 
On Android it works flawlessly. 

This error occurs regardless of which instantiation I use:
```
const bleManager = React.useRef(new BleManager())
let bleManager = new BleManager()
```
and on both debug and release builds. Haven't been able to get it to work for IOS.

Here is a shortened version of my bluetooth screen implementation

```
export default function BluetoothDeviceDashboard(navigation) {

  const bleManager = new BleManager();

  const [isScanning, setIsScanning] = React.useState(false);
  const [statusMessage, setStatusMessage] = React.useState('');
  const [foundDevice, setFoundDevice] = React.useState<Device>()

  const scanForDevices = () => { ... }
  const connectToDevice = (device: Device) => { ... }

  React.useEffect(() => {
    if (foundDevice) {
      setTimeout(() => { connectToDevice(foundDevice) }, 1500)
    }
  }, [foundDevice])

  const startScan = () => {
    setFoundDeviceInformation(undefined)
    const subscription = bleManager.onStateChange((state) => {
      if (state === 'PoweredOn') {
        scanAndConnect();
        subscription.remove();
      } else {
        navigation.goBack()
        Alert.alert('Bluetooth is off', 'Please turn on Bluetooth to continue')
      }
    }, true);
  }

  React.useEffect(() => {
    return () => {
      bleManager.destroy()
    }
  }, [])

  return (
    <View style={styles.container}>
      <View style={styles.headerContainer}>
        <Pressable onPress={() => { navigation.navigate('BluetoothRegisteredDevices') }}>
          <Text style={styles.linkText}>View All Registered Devices</Text>
        </Pressable>
       </View>
      <View style={styles.scanningContainer}>
        <Text style={styles.title}>{statusMessage}</Text>
        <ActivityIndicator size="large" color={Color.blue} animating={isScanning} />
      </View>
      <View style={styles.deviceContainer}>
        {foundDevice && foundDevice.name ? (
          <FoundBluetoothDeviceCard deviceInformation={getConfigData(foundDevice.name) as BluetoothDeviceInformation} activeReadings={liveReadings} />
        ) : null}
      </View>
      <View style={styles.buttonFooterContainer}>
        <Pressable style={[styles.buttonContainer, styles.defaultButton, isScanning ? styles.greyed : null]} onPress={() => startScan()} disabled={isScanning}>
          <Text style={styles.buttonText}>Clear and Scan Again</Text>
        </Pressable>
      </View>
    </View>
  )
}

Any help would be greatly appreciated, thanks

loewj-reekon, 2022-09-14:
What I ended up doing was instantiating `BleManager` in the root of my app. There's an example showing how to do this [here](https://github.com/mciechanowicz/react-native-ble-plx/tree/example-app/example). Afterwards, I was no longer seeing that error in Xcode. Good luck!

nepinney, 2022-09-14:
> What I ended up doing was instantiating/exporting `BleManager` in the root of my app (`index.js`). There's an example showing how to do this [here](https://github.com/mciechanowicz/react-native-ble-plx/tree/example-app/example). Afterwards, I was no longer seeing that error in Xcode. Good luck!

Love the work around, thank you!

dominik-czupryna-withintent, 2023-10-02:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-10-02 by dominik-czupryna-withintent