dotintent/react-native-ble-plx #894: Question

renatoramossilva, 2021-08-24:
- [ X] I am running the latest version
- [ X] I checked the documentation and found no answer
- [ X] I checked to make sure that this issue has not already been filed
- [ X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general.

## Question

I'm trying to unpair the ble device from the phone.

i.e. I can scan, connect the ble device. It´s OK! No news for now. I know that there is the cancelConnection but this function cancel the connection between ble device-App (But the device is still in the paired device list in the smartphone).

I would like to know if there is any way to remove/unpair the ble device from the smartphone.

Thanks! Regards

marcoj, 2021-08-30:
Hi,
I have the same question
is there a way for the lib or an api to unpair a device from the system bluetooth settings?

Thanks

mciechanowicz, 2021-11-08:
Hi, 
You should be able to unpair the device in Bluetooth settings on your phone.

renatoramossilva, 2021-11-08:
Hi @mciechanowicz,

Yes, I can do it,  but I would like to know if it is possible to unpair the device using the ble library.

Is this possible or should I do it from phone bt setting?

mciechanowicz, 2021-12-07:
No, it is not possible to unpair the device using this library, so you have to do it manually.

As far as I know, I don't think iOS allows you to unpair your device through the app so it's not possible to implement.

closed 2022-01-10 by mciechanowicz