dotintent/react-native-ble-plx #896: Bug Report: Shows warning with React Native 0.65

Naturalclar, 2021-08-28:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

Calling new BleManager in react-native 0.65 will not show any warning

## Current Behavior

What is the current behavior?

Calling new BleManager in react-native 0.65 shows the following warning

```
new NativeEventEmitter()` was called with a non-null argument without the required `removeListeners` method
```

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Use `react-native-ble-plx` in a project using react-native v0.65+ on an Android device.
2. Run `new BleManager()`

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.2
* Platform: Android.
* Platform logs (logcat/XCode):
* JS logs: ``new NativeEventEmitter()` was called with a non-null argument without the required `removeListeners` method.`
* Contents of the `package.json` file:
	"react-native": "0.65.1",
    "react-native-ble-plx": "^2.0.2"
* Formatted code sample or link to a repository:

```js
const ble = new BleManager();
```

uloco, 2021-09-13:
Will this affect behaviour of the ble library at the moment? I'm wondering if this is just a warning or if I should revert the upgrade of react-native now completely ...

I tried to understand what needs to be changed but not quite sure to be honest.
Basically the react-native eventemitter now needs to be constructed with an instance where the removeListeners function is required to be present on it.

See https://github.com/facebook/react-native/commit/114be1d2170bae2d29da749c07b45acf931e51e2

Naturalclar, 2021-09-13:
@uloco 
honestly, I don't think it affects the behavior of this library at the moment.

a different react-native module just added a stub implementation off event listeners to prevent the warning, and I think this library can be applied the same way

ref: https://github.com/software-mansion/react-native-reanimated/pull/2316/files

askie, 2021-11-25:
+1, the same bug

closed 2021-12-07 by srgtuszy

mciechanowicz, 2021-12-07:
Hi,

@Naturalclar thanks for reporting the issue and fixing it. Your PR has been merged! 💪

PR #901 (2021-12-07): Removed warning for Event Listeners in RN 0.65
Fixes #896 

Added required methods for NativeEventListeners to not show warning in React Native 0.65+

See this PR for more detail: https://github.com/react-native-netinfo/react-native-netinfo/pull/487