dotintent/react-native-ble-plx #897: Obtaining MAC address in iOS?

t3db0t, 2021-09-17:
The documentation for the `Device` `id` parameter says:
```
Device identifier: MAC address on Android and UUID on iOS.
```
I'm trying to filter detected devices by manufacturer; is there a way to get the MAC address instead of the UUID on iOS?

Xenofono, 2021-09-25:
Unfortunately I belive iOS doesn't expose MAC addresses for security reasons so you'll need to find some work around.

Tadejo9, 2021-09-25:
You could read serial number from the characteristic if you need the same identifier for both platforms.

tinnovia, 2021-09-27:
> You could read serial number from the characteristic if you need the same identifier for both platforms.

How can we achieve that, can you please describe explicitly? I can see 'characteristicUUID' inside console log of device from the function startDeviceScan() at _errorCodesToMessagesMapping.501 but, I don't know how to reach to that data. By the way variables (manufacturerData, overflowServiceUUIDs, rssi, serviceData, serviceUUIDs, solicitedServiceUUIDs, txPowerLevel etc.) returns null most of the time. Might I be doing something wrong?

Tadejo9, 2021-09-27:
The easiest way to see if you can access serial number would be the following:

```
connect to device
loop services
    loop characteristics
        is characteristic readable
        if yes; read characteristic value; parse value to readable format; check if any of parsed data is your serial number
```

Sometimes calling read method might not be enough. Specific devices have custom primary services and they are not part of the standard. They enable you to communicate with the device(erase data, transfer all data, setTime, ...). In all cases I've encountered serial number was either not present or accessible via read method.

tinnovia, 2021-09-27:
When I console log the device after connecting it, I don't see services. Documentation also doesn't have services in <Device> which is what bleManager.connectToDevice() function's only return.

Tadejo9, 2021-09-27:
I believe your claim is false.

You can check library documentation [here](https://dotintent.github.io/react-native-ble-plx/#device). There you should find services method.

tinnovia, 2021-09-27:
Yes, you are right. Didn't think services could be another method, that's. on me. :) https://github.com/dotintent/react-native-ble-plx/wiki/Device-Service-Discovery It's also detailed here.

mciechanowicz, 2021-12-07:
Hi.
Thanks to @Tadejo9 for the solution 🎉

I think the answer is sufficient so, I am closing the issue.

closed 2021-12-07 by mciechanowicz