dotintent/react-native-ble-plx #899: Xcode 13 error builds

nriccar, 2021-09-22:
Hello,

I noticed my Xcode updated to 13 
[image]

And right after that I started getting this error at build:

[image]
[image]
[image]
[image]
[image]

I run pod update 
```
Downloading dependencies
Installing MultiplatformBleAdapter 0.1.7 (was 0.1.6)
Installing react-native-ble-plx 2.0.2 (was 2.0.1)
Generating Pods project
Integrating client project
Pod installation complete! There are 81 dependencies from the Podfile and 71 total pods installed.
```

But same error 

Any guess?

wildseansy, 2021-09-22:
Seems like they're working on a fix: https://github.com/dotintent/MultiPlatformBleAdapter/commit/669e438acb384eee74f3f1ada3ba259cf7b7965b

but appears it hasn't been released yet

nriccar, 2021-09-22:
Thanks for the heads up. Ill check that one!

xanderdeseyn, 2021-09-27:
Seeing this as well, any idea when they will release the fix?

diogoviannaaraujo, 2021-09-28:
This has made impossible to develop on iOS 15 since XCode 12 does not accept iOS 15 devices as a development target.

wildseansy, 2021-09-28:
Maybe it wasn't clear, but can be fixed while they prepare release through patch package

https://github.com/dotintent/MultiPlatformBleAdapter/issues/79

> This has made impossible to develop on iOS 15 since XCode 12 does not accept iOS 15 devices as a development target.

LucasHimelfarb, 2021-09-28:
Hi @wildseansy sorry to bother you but could you indicate what are the steps to follow to solve the error? Reading and replicating what was said in the thread you shared, I can't solve the problem.

wildseansy, 2021-09-28:
1. Install `patch-package`
2. 
> Hi @wildseansy sorry to bother you but could you indicate what are the steps to follow to solve the error? Reading and replicating what was said in the thread you shared, I can't solve the problem.

wildseansy, 2021-09-28:
1. Install `patch-package` - `npm i patch-package --save-dev`
2. Add this under patches `patches/react-native-ble-plx+2.0.2.patch` 
```diff
diff --git a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
index def210f..0a92dcc 100644
--- a/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
+++ b/node_modules/react-native-ble-plx/react-native-ble-plx.podspec
@@ -17,5 +17,5 @@ Pod::Spec.new do |s|
   s.compiler_flags = '-DMULTIPLATFORM_BLE_ADAPTER'

   s.dependency 'React'
-  s.dependency 'MultiplatformBleAdapter', '0.1.7'
+  s.dependency 'MultiplatformBleAdapter', '0.1.9'
 end
```
3. Add `yarn patch-package` to your "postinstall" script in your `package.json`
> Hi @wildseansy sorry to bother you but could you indicate what are the steps to follow to solve the error? Reading and replicating what was said in the thread you shared, I can't solve the problem.

diogoviannaaraujo, 2021-09-29:
@wildseansy I've been trying doing the patch on postinstall but when running `pod install` I keep getting this error:

```
[!] CocoaPods could not find compatible versions for pod "MultiplatformBleAdapter":
  In Podfile:
    react-native-ble-plx (from `../node_modules/react-native-ble-plx`) was resolved to 2.0.2, which depends on
      MultiplatformBleAdapter (= 0.1.9)

None of your spec sources contain a spec satisfying the dependency: `MultiplatformBleAdapter (= 0.1.9)`.

You have either:
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.
```

wildseansy, 2021-09-29:
@diogoviannaaraujo - I forgot [step 4](https://github.com/dotintent/react-native-ble-plx/issues/899#issuecomment-929490351). Update your podfile to point to new cocoapod and you'll be good to go

> 4. Add this to your Podfile

diogoviannaaraujo, 2021-09-29:
Great, thanks that worked.

marcoj, 2021-09-30:
And you'll probably need to
```
pod update
pod repo update
pod install
```

LucasHimelfarb, 2021-10-01:
adding patch for podspec and also adding as pod bleAdapter is the working solution for now. Thanks for the help

Matheuschn, 2021-10-05:
It looks like the MultiPlatformBleAdapter library has already released version 0.1.9 with the fix. If possible, can we get a new react-native-ble-plx version with these fixes? Thanks!

The way i fixed it, at least until the fixes get to Cocoapods, is by replacing the library's folder (Pods/MultiplatformBleAdapter) with a clone from the 0.1.9 release (make sure the names match). Then, run 'pod install' and it should work for now.

LucasHimelfarb, 2021-12-02:
2.0.3 dont work for me

mciechanowicz, 2021-12-07:
Hi everyone! The issue with Xcode 13 was fixed in the 2.0.3 version!

@LucasHimelfarb can you give more details about your Issue?

mciechanowicz, 2021-12-10:
At this point, the problem seems to be solved so I close this issue. If anyone still has problems please report a new issue.

closed 2021-12-10 by mciechanowicz