dotintent/react-native-ble-plx #902: Unable to connect react native app to bluetooth.

AyushTripathi28, 2021-09-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ *] I am running the latest version
- [ *] I checked the documentation and found no answer
- [* ] I checked to make sure that this issue has not already been filed
- [* ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## I have been working on creating an app to connect Bluetooth(HC-05) with my React Native app. I am trying to make an IoT based hardware to monitor your ECG data(Heart Rate sensor AD8232) 

Need some good approaches and documents to achieve this.

Please describe the question.

mciechanowicz, 2021-11-04:
Hi @AyushTripathi28
Your's (HC-05) use BLE or classic Bluetooth?
If it's classic Bluetooth there is in the documentation that this library does not support classic Bluetooth devices

robertchoi, 2022-01-21:
please use "react-native-bluetooth-serial"

closed 2022-01-25 by mciechanowicz