dotintent/react-native-ble-plx #903: Loosing data sent from a BLE device

david-gettins, 2021-09-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

All data received from a GATT server is accessible.

## Current Behavior

With high update rates some data is missing. Not sure if it is due to bridge communication or too many callbacks queued or something else. This also happens at lower update rates but more characteristics notifying.

I have tried a few different approaches e.g. write staright to MMKV storage or buffer in an array then periodically save to storage.

Below are ECG (EKG) traces generated from a health and fitness device with an update rate of 2048Hz.

This is from an RN app using this library:
[image]

This is from a native app:
[image]

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Get a BLE device / GATT server that has a characteristic that emits data >=2048Hz.
2. Start monitoring the characteristic - storing in some kind of store
3. Plot the recieved data on a chart for analysis.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.2
* Platform: Android and iOS.
* Platform logs (logcat/XCode):
[react-native-ble-plx-issue-logs.txt](https://github.com/dotintent/react-native-ble-plx/files/7253062/react-native-ble-plx-issue-logs.txt)
* JS logs:
* Contents of the `package.json` file:
_Part of an NX monorepo so here is the app's linked dependencies:_
```json
{
  "name": "hardware-mobile",
  "version": "2.0.0",
  "private": true,
  "dependencies": {
    "@react-native-community/hooks": "*",
    "@react-native-community/netinfo": "*",
    "@react-native-firebase/analytics": "11.0.0",
    "@react-native-firebase/app": "11.0.0",
    "@react-native-firebase/iid": "11.0.0",
    "@react-navigation/core": "*",
    "@react-navigation/native": "*",
    "@react-navigation/stack": "*",
    "@gorhom/bottom-sheet": "*",
    "@shopify/restyle": "*",
    "@voximplant/react-native-foreground-service": "*",
    "buffer": "*",
    "date-fns": "*",
    "pino": "*",
    "ramda": "*",
    "react": "*",
    "react-native": "*",
    "react-native-app-auth": "*",
    "react-native-background-upload": "*",
    "react-native-ble-plx": "*",
    "react-native-circular-progress": "*",
    "react-native-config": "*",
    "react-native-device-info": "*",
    "react-native-fs": "*",
    "react-native-gesture-handler": "*",
    "react-native-mmkv": "*",
    "react-native-multithreading": "*",
    "react-native-nfc-manager": "*",
    "react-native-nordic-dfu": "*",
    "react-native-permissions": "*",
    "react-native-progress": "*",
    "react-native-reanimated": "*",
    "react-native-safe-area-context": "*",
    "react-native-screens": "*",
    "react-native-skeleton-placeholder": "*",
    "react-native-splash-screen": "*",
    "react-native-svg": "*",
    "react-native-vector-icons": "*"
  }
}
```

* Formatted code sample or link to a repository:

**`ble.notify - simply wraps the usual function so the consumer doesn't need to know about the ble manager`**
```ts
export function notify(
  deviceId: string,
  serviceId: string,
  characteristicId: string,
  callback: (
    error: BleError | null,
    characteristic: Characteristic | null,
  ) => Promise<void> | void,
): Subscription {
  return bleManager.monitorCharacteristicForDevice(
    deviceId,
    serviceId,
    characteristicId,
    callback,
  );
}
```

**`start notifications and write to a raw data store`**
```ts
let index = 0;

const sub = ble.notify(deviceId, service, characteristic, (error, char) => {
  if (error && error.errorCode !== BleErrorCode.OperationCancelled) {
    logger.warn(error);
    return;
  }

   if (char && char.value) {
    rawData.push(char.uuid, startDate, char.value, index++);
  }
});

monitored.push({ sub, uuid: characteristic, date: startDate });
```

**`rawData.push - using react-native-mmkv for the storage module`**
```ts
export function push(
  uuid: string,
  date: string,
  value: string,
  index = 0,
): void {
  const key = `${uuid}::${date}::${index}`;
  storage.storage.set(key, value);
  storage.storage.set(`${uuid}::${date}::latest`, index);
}
```

snydersaurus, 2022-08-02:
Did you ever get to the bottom of this @david-gettins ?

david-gettins, 2022-08-03:
Yes, well sort of... On Android you can reduce the data loss by setting connection priority to high:

```ts
bleManager.requestConnectionPriorityForDevice(device.id, ConnectionPriority.High)
```

Although, I still found data loss but only at around 0.6%. This could be seen as acceptable for most cases but presents a bit of an issue if your trying to do something with ECG signals (or similar) e.g. R peak detection, as the missing data points could be your R peak.

I messed around for sometime, even created a new app with nothing going on but the data collection to no avail. I have now created my own native module with Android and iOS implementations that save raw data on the native side using cooroutines (Android) and a dispatch queue (iOS). Only the data I need for the UI gets parsed and sent across to React Native. With this set up I could reduce the number of calls across the RN bridge by sending an event with the UI relevant data every 1 second. This was sufficient for the UI and no raw data is lost.

snydersaurus, 2022-08-04:
Thanks - I had a hunch that is what you would say.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska