dotintent/react-native-ble-plx #904: Devices with Extended Advertising enabled are not visible in scan

cdcformatc, 2021-09-30:
## Expected Behavior

Scan should return devices with Extended Advertising

## Current Behavior

Scan shows devices with standard advertising, including scan response. Devices that have extended advertising enabled do not appear. 

### Steps to Reproduce

1. Turn on peripheral that utilizes extended advertisement
2. Ensure your device supports extended advertisement, some older phones and tablets do not even if they support bluetooth 5
3. Call startDeviceScan in react-native app
4. Verify that callback does not return your device
5. Open a Bluetooth scanning app like nRF Connect, device appears and extended advertisement data is availiable. 
5. Disable extended advertisement on device
6. verify startDeviceScan callback picks up the device

### Context

I am developing a device and a react-native app that are intended to talk to each other. I would like to use extended advertisement. After enabling extended advertisement in my device's firmware the device is visible to nRF Connect, but not to my react-native app using this library. 

* Library version: 2.0.2
* Platform: Android 11 on a Samsung Galaxy Tab A7

luhaimin, 2022-04-14:
I also encountered the same problem. How can I find it in NRF connect, but this library does not return, or is it incorrectly configured?

cdcformatc, 2022-04-15:
From what I understand there are features that are available in Bluetooth 5 that are optional and do not have to be supported for a device to be classified as Bluetooth 5-enabled. Extended Advertising is one of these optional Bluetooth 5 additions that not all devices support. 

You can get the "nRF connect" app from the Play store and find the "Device Information" menu option and it will tell you what BT5 features your device supports. The device I was using, the Galaxy Tab 7 should support this function but this library does not.

saty9, 2023-06-16:
There is a library between this one an underlying library that does the actual scan call the underlying one has been updated to support extended advertising scans. https://github.com/dariuszseweryn/RxAndroidBle/issues/774

The one in-between however appears to have been abandoned and has had no movement since 2021 https://github.com/dotintent/MultiPlatformBleAdapter/pulls?q=is%3Apr+is%3Aclosed so doesn't use the updated version that supports extended advertising.

PR #1118 (2023-10-16): feat: provide option to disable legacy scan for Android
Added parameters to disable legacy scan option for Android.

According to [android documentation](https://developer.android.com/reference/android/bluetooth/le/ScanSettings.Builder#setLegacy(boolean)) disabling legacy scan is needed to correctly support scan devices with extended advertising.

Resolves #904
closed 2023-10-16 by gmiszewski-intent