dotintent/react-native-ble-plx #912: Covert characteristic value to string.

ravirupareliya, 2021-10-21:
I am reading characteristic from nordic device and able to read is successfully.

Issue I am facing is to convert it to string or actual value. According to firmware developer nordic is broadcasting values as unsigned short(2 bytes) and at the react native side we will need to covert it to hexadecimal and from hexadecimal to decimal. Now I am bit confused with how does it read value properly in nRF Connect app but not with this library.

Here is response I am getting while reading characteristic.

deviceID: "D9:C2:B1:BA:85:6C"
id: 10
isIndicatable: false
isNotifiable: true
isNotifying: true
isReadable: true
isWritableWithResponse: true
isWritableWithoutResponse: false
serviceID: 9
serviceUUID: "0000180a-0000-1000-8000-00805f9b34fb"
uuid: "00002ac4-0000-1000-8000-00805f9b34fb"
value: "8wI="

ChaossAdept7, 2021-11-14:
@Matheuschn thanks, also faced similar need, your method worked fine

mciechanowicz, 2021-12-07:
Hi!

@Matheuschn thanks a lot for the solution! 💪
@ravirupareliya Is Matheuschn answer sufficient for you and does it solve your problem?

ravirupareliya, 2021-12-16:
Yes, that worked, thank you @Matheuschn

closed 2021-12-16 by ravirupareliya