dotintent/react-native-ble-plx #918: Graphing data from gps sensor with Arduino with react-native

Cans11, 2021-10-31:
Hello there. I made a connection to the arduino nano 33 ble sense card using the react-native-ble-plx library. There is a gps sensor connected to the development board and it can send some parameters to the phone right now. But I want to parse the data and show it on the screen in graphic form. how can I do that? I looked through the library files and the code. data is using separate service and characteristic uuid.

Component/CharacteristicCard.tsx ;

`import React, { useEffect, useState } from 'react';
import {StyleSheet,Text, View,} from 'react-native';
import { Characteristic, fullUUID, UUID } from 'react-native-ble-plx';
import { TouchableOpacity } from 'react-native-gesture-handler';
import { Base64 } from '../lib/base64';
import { LogBox } from 'react-native';

LogBox.ignoreLogs([
 'Non-serializable values were found in the navigation state',
 'Each child in a list should have a unique "key" prop.',
 'ERROR',
 
]);

type CharacteristicCardProps = {
  char: Characteristic;
  
};

const decodeBleString = (value: string | undefined | null): string => {
  if (!value) {
    return '';
  }
  return Base64.decode(value).charCodeAt(0);
 
};

const CharacteristicCard = ({ char}: CharacteristicCardProps) => {
  const [measure, setMeasure] = useState('');
  const [_descriptor, setDescriptor] = useState<string | null>('');

  useEffect(() => {
    // discover characteristic descriptors
    char.descriptors().then((desc) => {
      desc[0]?.read().then((val) => {
        if (val) {
          setDescriptor(Base64.decode(val.value));
        }
      
      });
    });
   
    char.monitor((err, cha) => {
      if (err) {
        console.warn('ERROR');
        return;
      }
      // each received value has to be decoded with a Base64 algorythm you can find on the Internet (or in my repository 😉)
      setMeasure(decodeBleString(cha?.value));
    });
  }, [char]);
 

  // write on a charactestic the number 6 (e.g.)
  const writeCharacteristic = () => {
    
    

    // encode the string with the Base64 algorythm
    char
      .writeWithResponse(Base64.encode('6'))
      .then(() => {
        console.log('Success');
      })
      .catch((e) => console.log('Error', e));

     
     
  };
   
    
   
  return (
    <View>
    <TouchableOpacity
      key={char.uuid}
    
      style={styles.container}
      onPress={writeCharacteristic}>
      <Text style={styles.measure}>{measure}</Text>
   
    
    </TouchableOpacity>
  
   
    
   
  
    </View>
  );

};

const styles = StyleSheet.create({
  container: {
    backgroundColor: 'white',
    marginVertical: 12,
    
    padding: 12,
  },
  measure: { color: 'red', fontSize: 24 },
  descriptor: { color: 'black', fontSize: 24 },
  yuruyus:{color:'red', fontSize:21},
});

export { CharacteristicCard };

`

Device.tsx;

`import { StackScreenProps } from '@react-navigation/stack';
import React, { useCallback, useEffect, useState } from 'react';
import { Text, ScrollView, Button, View, StyleSheet } from 'react-native';
import { Service } from 'react-native-ble-plx';
import { ServiceCard } from '../components/ServiceCard';
import { RootStackParamList } from '../Navigator';
import {Device} from 'react-native-ble-plx';
import { LogBox } from 'react-native';

LogBox.ignoreLogs([
 'Non-serializable values were found in the navigation state',
 'Each child in a list should have a unique "key" prop.'
]);

function DeviceScreen({
  route,
  navigation,
  
}: StackScreenProps<RootStackParamList, 'Device'>) {
  // get the device object which was given through navigation params
  const { device } = route.params;

  const [isConnected, setIsConnected] = useState(false);
  const [services, setServices] = useState<Service[]>([]);

  // handle the device disconnection
  let disconnectDevice = useCallback(async function () {
      navigation.navigate('Home');
      const isDeviceConnected = await device.isConnected();
      if (isDeviceConnected) {
        await device.cancelConnection();
      }
    }, [device, navigation]);

  useEffect(() => {
    const getDeviceInformations = async () => {
      // connect to the device
      const connectedDevice = await device.connect();
      setIsConnected(true);
    

      // discover all device services and characteristics
      const allServicesAndCharacteristics = await connectedDevice.discoverAllServicesAndCharacteristics();
      // get the services only
      const discoveredServices = await allServicesAndCharacteristics.services();
      setServices(discoveredServices);
    };

    getDeviceInformations();
    
  

    // give a callback to the useEffect to disconnect the device when we will leave the device screen
    return function () {
      disconnectDevice();
    };
  }, [device, disconnectDevice]);

  return (
    <ScrollView contentContainerStyle={styles.container}>
      <Button title="Çıkış yap"  onPress={disconnectDevice}/>
      <View>
        <View style={styles.header}>
        <Text>{`Id : ${device.id}`}</Text>
          <Text>{`Name : ${device.name}`}</Text>
          <Text>{`Is connected : ${isConnected}`}</Text>
          <Text>{`RSSI : ${device.rssi}`}</Text>
          <Text>{`Manufacturer : ${device.manufacturerData}`}</Text>
          <Text>{`ServiceData : ${device.serviceData}`}</Text>
          <Text>{`UUIDS : ${device.serviceUUIDs}`}</Text>
        </View>
        {/* Displays a list of all services */}
        {services &&
          services.map((service) => <ServiceCard service={service} />)}
      </View>
    </ScrollView>
  );
}

const styles = StyleSheet.create({
  container: {
    padding: 12,
  },

  header: {
    backgroundColor: 'white',
    marginBottom: 12,
    borderRadius: 16,
    shadowColor: 'rgba(60,64,67,0.3)',
    shadowOpacity: 0.4,
    shadowRadius: 10,
    elevation: 4,
    padding: 12,
  },
});

export { DeviceScreen };
 

`

the source I got the sample from,
[https://github.com/Spoutnik97/react-native-arduino-ble-example](url)

Cans11, 2022-03-28:
yes I'm taking it with ble. I don't know what to do next

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska