dotintent/react-native-ble-plx #921: react-native-ble-plx 2.0.3 error

xuyushiguang, 2021-11-09:
[!] CocoaPods could not find compatible versions for pod "MultiplatformBleAdapter":
  In Podfile:
    react-native-ble-plx (from `../node_modules/react-native-ble-plx`) was resolved to 2.0.3, which depends on
      MultiplatformBleAdapter (= 0.1.9)

None of your spec sources contain a spec satisfying the dependency: `MultiplatformBleAdapter (= 0.1.9)`.

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

xuyushiguang, 2021-11-09:
who may help me?

Etchelon, 2021-11-09:
I had a similar issue and changed `0.1.7` to `0.1.9` in Podfile.lock

mciechanowicz, 2021-12-07:
Hi @xuyushiguang

I am sorry that you had a problem while using the library. It has already been resolved in this PR: https://github.com/dotintent/react-native-ble-plx/pull/928 and should be merged soon.

if you want to avoid this problem you can use the changes from the PR I pasted above, or just delete your Podfile.lock file and then try to install pods again 🙂

closed 2021-12-08 by mciechanowicz