dotintent/react-native-ble-plx #926: Is there any problem if i maintain instance from start to end?? like memory

whtjs, 2021-11-19:
i know that i have to use just one instance at time. But i have some questions about using pattern

is it ok that i create instance when app mount and use it at whole life cycle? or it will cause some issue so i have to create instance when i need it and after that  destroy and recreate when i need?

JoaoOliveiraRocha, 2022-01-26:
Hello @whtjs

From personal experience, there should be no problem if the connection to the device never breaks.

You can even use sometimes the instance to reconnect: however this is not reliable and I suggest when reconnecting, search for the same ID / name of the device to create a new connection.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska