dotintent/react-native-ble-plx #927: Question: is the configuration still the same since this package got added to expo? cuz im getting "undefined is not an object (evaluating '_BleModule.BleModule.createClient')"

yamil-90, 2021-11-19:
hello newbie here, 

i saw that there are similar questions regarding this error and it seems to be a configuration issue, however i followed the instructions several times and with 2 or 3 different projects and i got the same error in all of them.
so my question is if the doccumentation is updated since according to this post https://expo.canny.io/feature-requests/p/bluetooth-1, this package is now "sorta" supported by expo. 
do we still have to eject expo? as it is not mentioned in the post as a requirement

thanks for your help

mciechanowicz, 2021-11-29:
Hi @yamil-90 ✋

Have you tried using this: https://github.com/expo/config-plugins/tree/master/packages/react-native-ble-plx
If not, you can try to use these without having to eject expo.

Also, you can check that topic to be familiar with what's going on there: https://github.com/dotintent/react-native-ble-plx/pull/842

dashcraft, 2021-12-10:
@mciechanowicz maybe we could also get some movement on the package side to get something that's working into the repository?

Is this repo being actively maintained?

yamil-90, 2021-12-10:
@mciechanowicz in the end I decided to avoid all the trouble and did not use expo for this app :P got it working without it. thanks anyways, i will try it next time. 
@dashcraft  seems like the repo got updated recently

mciechanowicz, 2021-12-13:
@@dashcraft for some time there was a silence related to the change of library owner, but recently we started working on its further development.

@yamil-90 got it. so I am gonna close this issue now ;)

closed 2021-12-13 by mciechanowicz