dotintent/react-native-ble-plx #931: Does not scan when pass UUIDs to startDeviceScan on Android

IvKr95, 2021-12-03:
## Prerequisites

- [yes] I am running the latest version
- [yes] I checked the documentation and found no answer
- [no] I checked to make sure that this issue has not already been filed

## Expected Behavior

Runs the same on ios and android

## Current Behavior

iOS - gets packages when pass in UUIDs
Android - does not get packages when pass in UUIDs

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

* react: 16.13.1
* react-native: 0.63.4
* react-native-ble-plx: 2.0.3
* Platform: Android/iOS.

```js
bleManager.startDeviceScan([SERVICE_DATA], null, async (error: BleError | null, device: Device | null) => {
                if (error) {
                    console.log('SCAN ERROR =', error);
                }

                if (device) {
                }
            });
```

mciechanowicz, 2021-12-07:
Hi, can you specify on which device you are having this problem and what version of android?

IvKr95, 2021-12-08:
@mciechanowicz This is Samsung Galaxy A51 and Android version 11.

remipou, 2022-06-06:
having the same issue, can't pas an array of service UUIDs on Android. Galaxy S20 FE, Android version 12.

closed 2023-10-02 by dominik-czupryna-withintent