dotintent/react-native-ble-plx #932: Question

kjprice1, 2021-12-07:
## Question

Using this package, it will run on my iPhone if downloaded through xCode - so I know it works, but once deployed and testing through TestFlight it gets stuck on the splash screen and will not load. I am only using the ble manager to scan for nearby devices.

mciechanowicz, 2021-12-08:
Hi, can you provide some more information:
what version of the library you are using?
what version of react native do you have in your project?

It hangs on the splash screen(i.e. your application name is displayed or something custom if you've got a custom splash screen)? Or maybe something else is displayed and the app hangs on that?

Have you got any bug trackers for your app? Are there some logs about this issue?
Does nothing change after restarting the application?

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-10-06 by abursztynowska