dotintent/react-native-ble-plx #939: Device services became null and not discoverable after connection.

nico1996it, 2021-12-17:
Hello, i'm trying to interface a Bluetooth Qr code scanner with an iPad Air 2. 

```
function scanAndConnect() {
  manager.startDeviceScan(null, null, (error, device) => {
    if (error) {
      console.log(error);
      return;
    }

    if (device.name === 'C barcode scanner') {
      // Stop scanning as it's not necessary if you are scanning for one device.

      manager.stopDeviceScan();
      console.log(device);
      device
        .connect()
        .then((device) => {
          console.log(device);

          return device.services();
        })
        .then((device) => {
          console.log(device);
        });
    }
  });
}
```

This is the output of the first log 

```
Device {
id: "A6C3CCEA-55F5-53C1-5762-540E7E5ABE18"
isConnectable: true
localName: "C barcode scanner"
manufacturerData: null
mtu: 23
name: "C barcode scanner"
overflowServiceUUIDs: null
rssi: -82
serviceData: null
serviceUUIDs: ['0000fff0-0000-1000-8000-00805f9b34fb']
solicitedServiceUUIDs: null
txPowerLevel: null
}
```
Here everything is fine, and you can see the serviceUUIDs array 

This the log after the connection
```
Device{
id: "A6C3CCEA-55F5-53C1-5762-540E7E5ABE18"
isConnectable: null
localName: null
manufacturerData: null
mtu: 23
name: "C barcode scanner"
overflowServiceUUIDs: null
rssi: null
serviceData: null
serviceUUIDs: null
solicitedServiceUUIDs: null
txPowerLevel: null
}
```

As you can see, there is no serviceUUIDs array anymore.

The result of the third log is an empty array. There is basically no way to find the services after i'm connected to the device. 

This is the output of BLE scanner
[image]
From that app i can easily see two services and read the data when i scan something with the device. 

Thanks

mciechanowicz, 2021-12-29:
Hi,

Thank you for reporting the issue, we will try to take care of it.

Matheuschn, 2022-01-04:
Hello,

The information about the RSSI, the Manufacturer Data and the Service UUIDs are only available during the scanning process. Since the second log is getting the device object from the [.connect()](https://dotintent.github.io/react-native-ble-plx/#deviceconnect) method, all these properties are expected to be null. For more information, see issue #346.

You have two options:

- Save the serviceUUIDs array into a variable after scan;
- On the device object returned by the [.connect()](https://dotintent.github.io/react-native-ble-plx/#deviceconnect) method, call the [discoverAllServicesAndCharacteristics()](https://dotintent.github.io/react-native-ble-plx/#devicediscoverallservicesandcharacteristics) method and then call the [services()](https://dotintent.github.io/react-native-ble-plx/#deviceservices) method to get the array of Service objects, which will contain the UUIDs.

mciechanowicz, 2022-01-05:
@Matheuschn thanks for the answer!
I think the answer should dispel any doubts.

Closing for now. Please re-open that if you will have any further problems 😉

closed 2022-01-05 by mciechanowicz