dotintent/react-native-ble-plx #943: Background mode in Android

amduffy19, 2022-01-06:
Do you have to do something special to get it to work in the background for Android? I'm monitoring a heart rate characteristic which puts out a new value every second. When the screen is on, it's being received every second (roughly) but as soon as I turn the screen off it slows down to every 1.5 - 2.5 seconds. After a few minutes I turn the screen on and a ton of pending callbacks get executed all at once, which freezes the app. Am I doing something wrong? Is there a way to fix this? Thanks.

david-gettins, 2022-01-11:
Start a foreground service on Android and it should keep your app alive. I have used this [voximplant/react-native-foreground-service](https://github.com/voximplant/react-native-foreground-service) in a heart rate monitor app.

amduffy19, 2022-01-11:
@david-gettins Thank you. Does it just work as long as there is any foreground service running, or does the foreground service actually need to be running the BLE monitoring code? With that library I don't see a way to tell it what code to run in the foreground.

david-gettins, 2022-01-11:
Just so long as a foreground service is running. 

I did some testing a year or so ago and I successfully recorded a hr monitor for 4 hours while the app was backgrounded.

amduffy19, 2022-01-11:
@david-gettins Thank you! At least from initial testing it seems to be working! Really appreciate the help.

closed 2022-01-11 by amduffy19

lucaswitch, 2023-02-06:
It would be nice if this lib supports foreground services.