dotintent/react-native-ble-plx #946: Current react native app cannot build due to jcenter connection issue.

anunes9, 2022-01-17:
## Current Behavior

Current react native app cannot build due to jcenter connection issue.

### Context

* Library version: 2.0.3
* Platform: Android
* Platform logs (logcat/XCode):
* JS logs:

```
A problem occurred configuring project ':react-native-ble-plx'.
> Could not resolve all artifacts for configuration ':react-native-ble-plx:classpath'.
   > Could not resolve com.android.tools.build:builder:3.1.4.
     Required by:
         project :react-native-ble-plx > com.android.tools.build:gradle:3.1.4 > com.android.tools.build:gradle-core:3.1.4
      > Could not resolve com.android.tools.build:builder:3.1.4.
         > Could not get resource 'https://jcenter.bintray.com/com/android/tools/build/builder/3.1.4/builder-3.1.4.pom'.
            > Could not GET 'https://jcenter.bintray.com/com/android/tools/build/builder/3.1.4/builder-3.1.4.pom'.
               > Read timed out
```

On other libraries the solution was to replace **jcenter()** with **mavenCentral()** on build.gradle and it worked.

Any idea how to solve it?

Thanks

voids, 2022-01-19:
same issue.

mciechanowicz, 2022-01-19:
Hi.

We've got this already changed here: [android updates](https://github.com/dotintent/react-native-ble-plx/pull/920)
It should be merged in a few days now.

anunes9, 2022-01-19:
@mciechanowicz Thanks, waiting for merge

anunes9, 2022-03-15:
Hey guys, we are getting this error recurrently on our build process and it's related with this issue.
Do you have an estimate on when the pull request will be merged?

```
Could not determine the dependencies of task ':app:mergeReleaseAssets'.
> Could not resolve all dependencies for configuration ':app:releaseRuntimeClasspath'.
   > Could not determine artifacts for com.github.Polidea:MultiPlatformBleAdapter:b0272c0fae040cc90f790481c54e8f2d45304339: Skipped due to earlier error

```

Rain-Pavilion, 2022-06-07:
same issue.

Rain-Pavilion, 2022-06-07:
`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "21.4.7075529"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.2.2")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenCentral()
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }

        google()
        maven { url 'https://www.jitpack.io' }
    }
}
`

Rain-Pavilion, 2022-06-07:
> same issue.

`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "21.4.7075529"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.2.2")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenCentral()
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }

        google()
        maven { url 'https://www.jitpack.io' }
    }
}
`

dominik-czupryna-withintent, 2023-10-02:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-10-02 by dominik-czupryna-withintent